oliexdev/openScale #662: Crashing when measurements are received

voyager, 2020-12-01:
Since the 2.3.3-pro update the app crashes when bluetooth connection button is pressed.

Previously, tapping the BT button would receive the data from the scales and update the graph.

Now it crashes the app.

`Build version: 2.3.3-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2020-12-01 11:58:02 
Device: HUAWEI LYA-L29 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.NullPointerException: no valid service UUID provided
	at java.util.Objects.requireNonNull(Objects.java:228)
	at com.welie.blessed.BluetoothPeripheral.getCharacteristic(BluetoothPeripheral.java:1077)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:200)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:188)
	at com.health.openscale.core.bluetooth.BluetoothSenssun.synchroniseDate(BluetoothSenssun.java:199)
	at com.health.openscale.core.bluetooth.BluetoothSenssun.onNextStep(BluetoothSenssun.java:85)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:460)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.access$000(BluetoothCommunication.java:46)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$2.onConnectedPeripheral(BluetoothCommunication.java:370)
	at com.welie.blessed.BluetoothCentral$4$1.run(BluetoothCentral.java:251)
	at android.os.Handler.handleCallback(Handler.java:900)
	at android.os.Handler.dispatchMessage(Handler.java:103)
	at android.os.Looper.loop(Looper.java:219)
	at android.app.ActivityThread.main(ActivityThread.java:8347)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1055)`

oliexdev, 2020-12-02:
please provide the openScale log file

voyager, 2020-12-02:
Here's the debug log (with private data reducted).
Thanks Olie.

```
2020-12-02 12:27:51.575 Debug [1] AboutPreferences: Debug log enabled, openScale v2.3.3-pro (52), SDK 29, HUAWEI LYA-L29
2020-12-02 12:27:51.577 Debug [1] AboutPreferences: Selected user id(1) name(Mario) birthday(XXX XXX XX 00:00:00 GMT+03:00 XXXX) age(XX) body height(XXX.00) scale unit(kg) gender(male) initial weight(XX.00) goal weight(XX.00) goal date(Tue Oct 29 11:03:59 GMT+02:00 2019) measure unt(cm) activity level(1) assisted weighing(false)
2020-12-02 12:27:17.760 Debug [1] OpenScale: Trying to connect to bluetooth device [18:7A:93:41:E7:8B] (SENSSUN FAT)
2020-12-02 12:27:17.763 Debug [1] BluetoothCommunication: No location permission, connecting without LE scan
2020-12-02 12:27:18.765 Debug [1] BluetoothCommunication: Try to connect to BLE device 18:7A:93:41:E7:8B
2020-12-02 12:27:18.869 Info [1] BluetoothPeripheral: connect to 'SENSSUN FAT' (18:7A:93:41:E7:8B) using TRANSPORT_LE
2020-12-02 12:27:22.737 Info [32382] BluetoothPeripheral: connected to 'SENSSUN FAT' (BOND_NONE) in 3.9s
2020-12-02 12:27:22.739 Debug [1] BluetoothPeripheral: discovering services of 'SENSSUN FAT' with delay of 0 ms
2020-12-02 12:27:23.737 Info [32382] BluetoothPeripheral: discovered 9 services for 'SENSSUN FAT'
2020-12-02 12:27:23.740 Debug [1] BluetoothCommunication: connected to 'SENSSUN FAT'
2020-12-02 12:27:23.742 Debug [1] BluetoothCommunication: Step Nr 0
2020-12-02 12:27:23.743 Debug [1] BluetoothSenssun: Sync Date
2020-12-02 12:27:23.749 Debug [1] BluetoothCommunication: Invoke write bytes [A5 30 14 01 51 00 00 96 00] on null
```

oliexdev, 2020-12-02:
please grant access to the location permission for openScale and try again.

voyager, 2020-12-03:
Hey Ollie, yeah saw that about location when I was formatting the message, but got distracted with work before I could try it.
Anyway, I've turned on the location today and, surprise, it worked :)

Thanks for a quick help 🙏

closed 2020-12-03 by voyager