oliexdev/openScale #664: Renpho ES-24M (works but says "Select Device" then doesn't display anything)

famewolf, 2020-12-13:
**Describe the bug**
I attempted to use the ES-24M scale with openscale and it appears to work...at least it pulls the weight, bmi, bmr and a few others.  The issue comes when enabling bluetooth it says "Select Device" but then never shows anything but obviously it's finding the scale because after I weigh if I hit the + the most recent entry is there and I can choose to save it.  (is that how it's supposed to work?  Is there an option to auto save the entry from scale to phone without having to manually choose to save it?)

[image]

famewolf, 2020-12-13:
I was mistaken.   It's apparently pulling the info from the Renpho app somehow because when I uninstalled the app the weight never changed even it showed a lower one but the time did.

oliexdev, 2020-12-15:
please fillout the whole bug template and attach the openScale log file

famewolf, 2020-12-15:
It doesn't show up at all and the log just shows scan started and ended.  I figure the data came from the phone since no link to the scale appears to be established.  I'll close this out.  Sorry for the confusion.

closed 2020-12-15 by famewolf