oliexdev/openScale #667: Turn 'disconnect' button into a 'connect' button if connection failed

alexanderadam, 2020-12-16:
**Describe the bug**

Apart from [the resizing bug](https://github.com/oliexdev/openScale/issues/603) I got connection issues at one point (it's been a while since this started, though).
I'm somehow used to it: it takes a few runs until openScale is getting a connection.
If the connection dies, however, the button for connecting is still in the `connected` state. Thus I have to "disconnect" (although it is already disconnected) to retry connecting.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to connect
2. Wait to show the connection issue notification
3. Tap on the connect button again

Reproduced with: 2.3.3 (F-Droid

**Expected behavior**

Button should present the current state.

oliexdev, 2021-01-08:
it is in the state connection lost, which the symbol is grey and not in the initial state "not connected". After a connection lost it is better to disconnect once again before a new connection.

closed 2021-01-08 by oliexdev

alexanderadam, 2021-01-08:
> After a connection lost it is better to disconnect once again before a new connection.

Well, then should openScale disconnect, right?
Or what use does the state `connection lost` have for the user? :thinking: 

Semantically it doesn't make any difference for since the icon only only presents two states anyway, right?
The only difference seems to be the notification and that I have to 'disconnect' from 'connection lost' — which doesn't appear useful at all.

Imagine you're are calling someone from your phone and the connection is getting lost. Why should you then 'disconnect' from something that isn't there anymore anyway?

alexanderadam, 2024-04-20:
This bug is still present after all these years. :disappointed: