oliexdev/openScale #668: Scale Support Request: Renpho ES-24M Bluetooth Scale

famewolf, 2020-12-16:
Renpho ES-24M Bluetooth Scale
product page:
https://renpho.com/collections/scale/products/renpho-es-24m-bluetooth-body-fat-weight-scale
vendors app:
https://play.google.com/store/apps/details?id=com.qingniu.renpho

To support a new scale it is necessary to gather some information.

Step 1: Read the general reverse engineer process

I have read the "how to reverse engineer a Bluetooth 4.x wiki."  Since the scale is not detected I'm unsure how far I can get with this.  I would assume the process would be similar to the Renpho ES-26M which is supported by openscale.

2. Find out the Bluetooth services and characteristic
OpenScale dev version was not able to find my scale. The log shows the scan starting and ending with no info in between.
BLE scanner Raw Data Screenshot attached.

[image]

oliexdev, 2021-01-08:
You need to find the scale at least otherwise your might have a faulty scale. Does it work with the vendors app?

closed 2021-01-08 by oliexdev

famewolf, 2021-01-08:
It works fine with the vendor app and I made sure to step on the scale when searching for it in openscale.

oliexdev, 2021-01-09:
I added a new FAQ entry at https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#why-is-my-scale-not-found

guix77, 2021-09-02:
See #769

Androphin, 2022-10-25:
Same with model Renpho es-26m-w

Currently investigating.