oliexdev/openScale #669: Import scripts in FAQ and wiki

Martin1887, 2020-12-20:
This is a good app to migrate from other proprietary ones so having migration guides would be great for this purpose.

As example, I provide a script to migrate from MiFit 'BODY' file, but note that only weight and timestamp are migrated because I had not other statistics. The MiFit 'BODY' file header is included in the script for reference. I had also to remove a weird UTF-8 character at the start of the file before running this script.

```python
#!/usr/bin/python

import argparse
import csv
import datetime

OPENSCALE_HEADER = '"biceps","bone","caliper1","caliper2","caliper3","calories","chest","comment","dateTime","fat","hip","lbm","muscle","neck","thigh","visceralFat","waist","water","weight"'

_MIFIT_BODY_HEADER = 'timestamp,weight,height,bmi,fatRate,bodyWaterRate,boneMass,metabolism,muscleRate,visceralFat,impedance'

if __name__ == '__main__':
    parser = argparse.ArgumentParser()
    parser.add_argument('mifit_BODY_csv_file_path')
    parser.add_argument('output_path')
    
    args = parser.parse_args()
    
    with open(args.mifit_BODY_csv_file_path, 'r') as inp:
        reader = csv.DictReader(inp)
        with open(args.output_path, 'w') as outp:
            writer = csv.DictWriter(outp, OPENSCALE_HEADER.replace('"', '').split(','))
            outp.write(f'{OPENSCALE_HEADER}\n')
            
            for line in reader:
                timestamp = int(line['timestamp'])
                output_date = datetime.datetime.fromtimestamp(timestamp).strftime('%Y-%m-%d %H:%M')
                weight = line['weight']
                writer.writerow({'dateTime': output_date, 'weight': weight})
                
```

At least this script could be included in the FAQ and the wiki.

Regards.

oliexdev, 2021-01-08:
Thanks I added a new wiki page @ https://github.com/oliexdev/openScale/wiki/Useful-import-scripts

closed 2021-01-08 by oliexdev

dancaoma, 2023-05-25:
@oliexdev @Martin1887 
Today I have been trying to import all my data into Zepp with this wonderful script and I have come across the problem of the character that appears at the beginning of the parser processing.
With your permission, I have updated the script to be able to import correctly, skipping this annoying error.
I've also made a couple of additions that were missing from the original companion script.
Thanks for your work!
```
#!/usr/bin/python

import argparse
import csv
import datetime
import codecs

OPENSCALE_HEADER = '"biceps","bone","caliper1","caliper2","caliper3","calories","chest","comment","dateTime","fat","hip","lbm","muscle","neck","thigh","visceralFat","waist","water","weight"'

_MIFIT_BODY_HEADER = 'time,weight,height,bmi,fatRate,bodyWaterRate,boneMass,metabolism,muscleRate,visceralFat,impedance'

if __name__ == '__main__':
    parser = argparse.ArgumentParser()
    parser.add_argument('--mifit_BODY_csv_file_path', default='\\path-to-your-mifit-file\\mifit-file.csv')
    parser.add_argument('--output_path', default='\\path-to-your-output-file\\output-file.csv')
    
    args = parser.parse_args()
    
    with codecs.open(args.mifit_BODY_csv_file_path, 'r', encoding='utf-8-sig') as inp:
        reader = csv.DictReader(inp)
        with open(args.output_path, 'w') as outp:
            writer = csv.DictWriter(outp, OPENSCALE_HEADER.replace('"', '').split(','))
            outp.write(f'{OPENSCALE_HEADER}\n')
            
            for line in reader:
                if 'time' in line:
                    timestamp_str = line['time']
                    timestamp = datetime.datetime.strptime(timestamp_str, '%Y-%m-%d %H:%M:%S%z')
                    output_date = timestamp.strftime('%Y-%m-%d %H:%M:%S%z')
                    weight = line['weight']
                    bone = line['boneMass']
                    muscle = line['muscleRate']           
                    visceralFat = line['visceralFat']     
                    fat = line['fatRate']    
                    water = line['bodyWaterRate']       
                    calories = line['metabolism']                        
                    writer.writerow({'dateTime': output_date, 'weight': weight, 'bone': bone, 'muscle': muscle, 'visceralFat': visceralFat, 'fat': fat, 'water': water, 'calories': calories})
                else:
                    print(f"The line does not have a 'time' field")
```

At least this updated script could be included in the FAQ and the wiki.

Regards.

yusufani, 2023-11-07:
Thanks for the code @danicosas. I have encountered with null value problem, so updated the code slightly for Mi body scale 2. 

Note: I don't have impedance value, I don't know the reason.
```

#!/usr/bin/python

import argparse
import csv
import datetime
import codecs

OPENSCALE_HEADER = '"biceps","bone","caliper1","caliper2","caliper3","calories","chest","comment","dateTime","fat","hip","lbm","muscle","neck","thigh","visceralFat","waist","water","weight"'

_MIFIT_BODY_HEADER = 'time,weight,height,bmi,fatRate,bodyWaterRate,boneMass,metabolism,muscleRate,visceralFat'

if __name__ == '__main__':
    parser = argparse.ArgumentParser()
    parser.add_argument('--mifit_BODY_csv_file_path', default='BODY_xxxxx.csv')
    parser.add_argument('--output_path', default='export.csv')
    
    args = parser.parse_args()
    
    with codecs.open(args.mifit_BODY_csv_file_path, 'r', encoding='utf-8-sig') as inp:
        reader = csv.DictReader(inp)
        with open(args.output_path, 'w') as outp:
            writer = csv.DictWriter(outp, OPENSCALE_HEADER.replace('"', '').split(','))
            outp.write(f'{OPENSCALE_HEADER}\n')
            
            for line in reader:
                if 'time' in line:
                    timestamp_str = line['time']
                    timestamp = datetime.datetime.strptime(timestamp_str, '%Y-%m-%d %H:%M:%S%z')
                    output_date = timestamp.strftime('%Y-%m-%d %H:%M:%S%z')
                    weight = line.get('weight', '').replace('null', '')
                    bone = line.get('boneMass', '').replace('null', '')
                    muscle = line.get('muscleRate', '').replace('null', '')           
                    visceralFat = line.get('visceralFat', '').replace('null', '')     
                    fat = line.get('fatRate', '').replace('null', '')    
                    water = line.get('bodyWaterRate', '').replace('null', '')       
                    calories = line.get('metabolism', '').replace('null', '')
                    writer.writerow({'dateTime': output_date, 'weight': weight, 'bone': bone, 'muscle': muscle, 'visceralFat': visceralFat, 'fat': fat, 'water': water, 'calories': calories})
                else:
                    print(f"The line does not have a 'time' field")
```

danicosas, 2023-11-08:
Thank you, @yusufani . 
I opened a bug by commenting on another issue in the following link, and the original developer ignored me. 
Let's see if a more proactive developer knows why.

Martin1887, 2023-11-10:
> Thank you, @yusufani . I opened a bug by commenting on another issue in the following link, and the original developer ignored me. Let's see if a more proactive developer knows why.

Which issue? Sorry if I ignored anything, I have only received a notification from a previous comment in this issue and I only understood from it that you updated successfully the script.

danicosas, 2023-11-11:
https://github.com/oliexdev/openScale/issues/962

Martin1887, 2023-11-11:
I had not seen it but no idea anyways, I only imported basic data.

MiceX, 2026-08-27:
And another one update

```python
#!/usr/bin/python3

import argparse
import csv
import datetime
import codecs

OPENSCALE_HEADER = '"DATE","TIME","BCM","BICEPS","BMI","BMR","BODY_FAT","BONE","CALIPER","CALIPER_1","CALIPER_2","CALIPER_3","CALORIES","CHEST","COMMENT","ECW","HEART_RATE","HIPS","ICW","IMPEDANCE","IMPEDANCE_LOW","LBM","MUSCLE","NECK","PROTEIN","TDEE","THIGH","VISCERAL_FAT","WAIST","WATER","WEIGHT","WHR","WHTR"'

_MIFIT_BODY_HEADER = 'time,weight,height,bmi,fatRate,bodyWaterRate,boneMass,metabolism,muscleRate,visceralFat'

if __name__ == '__main__':
    parser = argparse.ArgumentParser()
    parser.add_argument('--mifit', default='BODY_xxxxx.csv')
    parser.add_argument('--output', default='export.csv')
    
    args = parser.parse_args()
    
    with codecs.open(args.mifit, 'r', encoding='utf-8-sig') as inp:
        reader = csv.DictReader(inp)
        with open(args.output, 'w') as outp:
            writer = csv.DictWriter(outp, OPENSCALE_HEADER.replace('"', '').split(','))
            outp.write(f'{OPENSCALE_HEADER}\n')
            
            for line in reader:
                if 'time' in line:
                    timestamp_str = line['time']
                    timestamp = datetime.datetime.strptime(timestamp_str, '%Y-%m-%d %H:%M:%S%z')
                    output_date = timestamp.strftime('%Y-%m-%d %H:%M')
                    weight = line.get('weight', '').replace('null', '')
                    bone = line.get('boneMass', '').replace('null', '')
                    bmi = line.get('bmi', '').replace('null', '')
                    muscle = line.get('muscleRate', '').replace('null', '')           
                    visceralFat = line.get('visceralFat', '').replace('null', '')     
                    fat = line.get('fatRate', '').replace('null', '')    
                    water = line.get('bodyWaterRate', '').replace('null', '')       
                    calories = line.get('metabolism', '').replace('null', '')
                    writer.writerow({
                        'DATE': timestamp.strftime('%Y-%m-%d'),
                        'TIME': timestamp.strftime('%H:%M:%S'),
                        'BMI': bmi,
                        'BODY_FAT': fat,
                        'BONE': bone,
                        'CALORIES': calories,
                        'MUSCLE': muscle,
                        'VISCERAL_FAT': visceralFat,
                        'WEIGHT': weight,
                        'WATER': water
                    })
                else:
                    print(f"The line does not have a 'time' field")
```