oliexdev/openScale #671: US Navy body fat estimate

jimr1603, 2020-12-22:
The formulas for estimating body fat percentage (BF%) from age, sex and weight are good, but I would like to improve accuracy by including waist (& hip, female) & neck measurements. 

I prefer this measure to calipers since I can get the same measurement twice in a row! 

The formulas aren't complicated - [this site](https://www.calculatorpro.com/calculator/us-navy-body-fat-calculator/) has them in inches & lbs. If you like the concept I'm happy to dig out the paper the USA DoD wrote about the method for the citations in the project. 

I'd offer to code it but I've never managed to get something to compile in Android Studio.

oliexdev, 2021-01-08:
Currently I used the caliper measurement formula by Jackson, Polloc but another formula could be implemented as an option in the caliper measurement settings in openScale like for the body water measurement. If you want to do it your are welcome to send a PR.

jimr1603, 2021-01-08:
I got the latest commit of the app to build on my PC, so I'll give it a
try.

>