oliexdev/openScale #673: Garmin connect sync

sebesvoj, 2020-12-29:
Hi, it's perfect app and it works fine, but if you'll add garmin connect sync, it will be great! I've read something about garmin API, and it seems it's possible. Now I have to use this app (https://github.com/kochka/WeightLogger), sync works great, but I have to use open scale, than manually transfer data (it to possible to import CSV in weight logger) and than sync it to garmin. It's only way o know. Thanks 😊

sebesvoj, 2020-12-29:
If I understand the code from weight logger correctly, it's just uploads file with weight data to https://connect.garmin.com/modern/import-data. No API, right?

oliexdev, 2021-01-08:
it is more a license issue with the Garmin license, which I don't accept for privacy reasons.

closed 2021-01-08 by oliexdev

pervolianinen, 2022-05-29:
What about adding open code for garmin sync?
github.com/tormoder/fit