oliexdev/openScale #674: Tutorial text is invisible when system theme is dark

str4d, 2020-12-29:
**Describe the bug**
The tutorial text is not readable when the system theme is set to dark mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Install openScale dev app.
2. Open for the first time.
3. Briefly see the main app's background is dark.
4. Tutorial pops up.
5. The tutorial's background is white, but the regular text is also white, so it is invisible. Images are visible, along with linkified text (which is blue).

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): Yes.

**Expected behavior**
I can read the tutorial, because either the tutorial background is dark per dark theme, or the text is dark per light theme.

oliexdev, 2021-01-08:
hmm this should be fixed in the latest release. Which Android version do you have? Could you attach a screenshot?

oliexdev, 2021-01-08:
Which openScale version do you use?

closed 2021-02-17 by oliexdev