oliexdev/openScale #676: Bluetooth connection issues with supported Xiaomi Mi scale v2

stepeck, 2020-12-29:
Hi,
I bought the Xiaomi scale for it is listed as fully supported.

Well, in the Bluetooth connection app of android i can scan and see the device but not been able to connect with the hint "connection only possible with an app".

In openscale the same issue: detection possitive but no connection possible.

So I installed BLE scanner and  detect the scale too even seeing data transfered.

What am I missing?

Do I have to install the original app?

Thanks!

oliexdev, 2021-01-08:
In openScale goto settings->Bluetooth search for the scale click on it and they click on the Bluetooth icon in openScale in the overview fragment.

closed 2021-01-08 by oliexdev

stepeck, 2021-01-11:
Hi,thats what I am doing. And this is the result:It is not possible to connect with the scale.With the original app everything is fine.Please check the Bluetooth name in the screenshot, maybe there is a new firmware/protocol?Yours sincerely

oliexdev, 2021-01-11:
The Bluetooth name is the same for mi scale v2.