oliexdev/openScale #677: Sanitas SBF70 has stopped syncing (ERROR: Failed writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>)

apinkney97, 2021-01-03:
# Summary
Hi, my Sanitas SBF-70 scales last synced successfully on 22nd November 2020. (Or at least that's the date of the latest measurement that synced.)

I've included the full debug log at the bottom of this comment, but I think these are the key lines of interest:
```
2021-01-03 17:22:10.014 Debug [2] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-01-03 17:22:15.097 Error [44522] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb> failed, status 'GATT_ERROR'
2021-01-03 17:22:15.100 Error [2] BluetoothCommunication: ERROR: Failed writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-01-03 17:22:15.101 Info [44522] BluetoothPeripheral: peripheral 'SANITAS SBF70' disconnected with status 'CONNECTION_TIMEOUT' (8)
2021-01-03 17:22:15.117 Debug [2] BluetoothCommunication: disconnected 'SANITAS SBF70' with status 8
```

This looks like it's the same symptoms as #503, though that says there was a fix in v2.1.2.

# Steps to reproduce
Start a bluetooth connection by disabling and then enabling bluetooth using the button in the top bar of the app.

The app appears to be connecting to the scale ok; it says
- `Connecting to: SANITAS SBF70`

then shortly afterwards
- `Connected`

then about a minute later
- `Bluetooth connection closed`.

During this time, the bluetooth symbol on the scale never lights up. (I'm fairly sure it used to when the display was on.)

# Things I have tried

- Replace the batteries in the scale
- Power cycle my phone
- Look for obvious things in the debug log

I'm using OpenScale Pro from the Google Play Store. I note there were [a couple of releases in November](https://github.com/oliexdev/openScale/tags) but I don't know when they were pushed to the Play Store. I also don't know exactly when the app updated on my phone, but I have auto updates enabled, so it should be fairly close to release time.

# Debug log
```
2021-01-03 17:21:49.637 Debug [2] AboutPreferences: Debug log enabled, openScale v2.3.5-pro (54), SDK 30, Google Pixel 5
2021-01-03 17:21:49.639 Debug [2] AboutPreferences: Selected user id(3) name(x) birthday(x) age(x) body height(x) scale unit(kg) gender(x) initial weight(x) goal weight(x) goal date(x) measure unt(cm) activity level(x) assisted weighing(false)
2021-01-03 17:22:05.194 Debug [2] OpenScale: Disconnecting from bluetooth device
2021-01-03 17:22:05.194 Debug [2] BluetoothCommunication: Bluetooth disconnect
2021-01-03 17:22:05.195 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-01-03 17:22:05.195 Error [2] BluetoothCentral: cannot cancel connection to unknown peripheral C4:BE:84:06:34:A1
2021-01-03 17:22:07.574 Debug [2] OpenScale: Trying to connect to bluetooth device [C4:BE:84:06:34:A1] (SANITAS SBF70)
2021-01-03 17:22:07.577 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2021-01-03 17:22:07.584 Info [2] BluetoothCentral: scan started
2021-01-03 17:22:07.585 Debug [2] BluetoothCommunication: Stop machine state
2021-01-03 17:22:07.644 Debug [2] BluetoothCommunication: Found peripheral 'SANITAS SBF70'
2021-01-03 17:22:07.647 Info [2] BluetoothCentral: scan stopped
2021-01-03 17:22:08.649 Debug [2] BluetoothCommunication: Try to connect to BLE device C4:BE:84:06:34:A1
2021-01-03 17:22:08.755 Info [2] BluetoothPeripheral: connect to 'SANITAS SBF70' (C4:BE:84:06:34:A1) using TRANSPORT_LE
2021-01-03 17:22:08.941 Info [44522] BluetoothPeripheral: connected to 'SANITAS SBF70' (BOND_NONE) in 0.2s
2021-01-03 17:22:08.943 Debug [2] BluetoothPeripheral: discovering services of 'SANITAS SBF70' with delay of 0 ms
2021-01-03 17:22:09.976 Info [44522] BluetoothPeripheral: discovered 5 services for 'SANITAS SBF70'
2021-01-03 17:22:09.978 Debug [2] BluetoothCommunication: connected to 'SANITAS SBF70'
2021-01-03 17:22:09.978 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2021-01-03 17:22:09.979 Debug [2] BluetoothCommunication: Resume machine state
2021-01-03 17:22:09.979 Debug [2] BluetoothCommunication: Step Nr 0
2021-01-03 17:22:09.979 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2021-01-03 17:22:09.980 Debug [2] BluetoothCommunication: Stop machine state
2021-01-03 17:22:09.985 Debug [2] MainActivity: Bluetooth connection successful established
2021-01-03 17:22:10.011 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2021-01-03 17:22:10.012 Debug [2] BluetoothCommunication: Resume machine state
2021-01-03 17:22:10.012 Debug [2] BluetoothCommunication: Step Nr 1
2021-01-03 17:22:10.012 Debug [2] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2021-01-03 17:22:10.013 Debug [2] BluetoothCommunication: Stop machine state
2021-01-03 17:22:10.014 Debug [2] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-01-03 17:22:15.097 Error [44522] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb> failed, status 'GATT_ERROR'
2021-01-03 17:22:15.100 Error [2] BluetoothCommunication: ERROR: Failed writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-01-03 17:22:15.101 Info [44522] BluetoothPeripheral: peripheral 'SANITAS SBF70' disconnected with status 'CONNECTION_TIMEOUT' (8)
2021-01-03 17:22:15.117 Debug [2] BluetoothCommunication: disconnected 'SANITAS SBF70' with status 8
2021-01-03 17:23:09.984 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect
2021-01-03 17:23:09.985 Debug [2] BluetoothCommunication: Bluetooth disconnect
2021-01-03 17:23:09.986 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-01-03 17:23:09.987 Error [2] BluetoothCentral: cannot cancel connection to unknown peripheral C4:BE:84:06:34:A1
2021-01-03 17:23:09.996 Debug [2] MainActivity: Bluetooth connection successful disconnected
2021-01-03 17:23:19.785 Debug [2] AboutPreferences: Debug log disabled
```

oliexdev, 2021-01-08:
> I'm using OpenScale Pro from the Google Play Store. I note there were a couple of releases in November but I don't know when they were pushed to the Play Store. I also don't know exactly when the app updated on my phone, but I have auto updates enabled, so it should be fairly close to release time.

Nothing changed for the SBF70 code, see the release notes.

It seems you cannot write any commands to the scale. Mostly the battery is low but as you replace the battery of the scale it another problem. Maybe reset the scale?

apinkney97, 2021-01-09:
I reset scale and it now syncs, though of course it deleted all the stored measurements, which I was trying to avoid.

I guess this must be a bug in the scale itself.

Thanks for your help!

closed 2021-01-09 by apinkney97

gnaghi, 2021-02-10:
I still have the exact same problem. 
I resetted the scale, reinstalled Openscale, tried to init my profile with HealthCoach (without success), and without (is it mandatory to use HealthCoach?).
Openscale shows «connected», but he scale doesn’t have the Bluetooth logo, like it usually did.

apinkney97, 2021-02-14:
@gnaghi after I reset my scale I did not reinstall OpenScale or use HealthCoach - I assume OpenScale pushed my details back to the scale when it reconnected.

gnaghi, 2021-02-22:
After 2 full reset, I had the same error as you, and after a second try, I got the same error reported here : #552.
My profile wasn’t pushed back into the scale.

gnaghi, 2021-07-21:
I’ve tested with the dev version, hoping that the #718 would fix this bug, but it is still here.
The user seems to be created, but then I have the nullpointer crash.