oliexdev/openScale #678: Used Scale -> data history from another person

jkopka, 2021-01-04:
**Is your feature request related to a problem? Please describe.**
Yes. I have bought a mi scale from another person. the App downloads all of his data.

**Describe the solution you'd like**
Maybe a setting option where this feature could be downloaded or limited to a date range or a start date.

oliexdev, 2021-01-08:
that's not possible (or at least I never found) a command to delete any data on the mi scale. To avoid this problem, that the data is always downloaded from the scale, you can just delete these data once in openScale. It will only be marked in openScale as deleted but is actually not deleted in the database. So these data will not shown again.

closed 2021-01-08 by oliexdev

CarlosGS, 2021-01-15:
I'm facing the same issue. Is there a way to bulk-delete these data from openScale? Choosing to "delete the database" makes the app grab all the obsolete historic again :-/