oliexdev/openScale #680: Problem With AcuHealth Scale

03172008, 2021-01-07:
It isn't receiving the correct lbs measurement from the AcuHealth scale. It connects perfectly fine on both ends, except that it shows lbs on the scale's end, and shows kg measurement as a lbs measurement on the user end. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect AcuHealth Scale to bluetooth connection with phone
2. Stand on scale and transfer measurement
3. Look at the lbs entry of the log
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): Yes. Problem Still exists.

**Expected behavior**
I expected it to use the lbs data of weight, and not the kgs.

**Debug log**
[openScale_2021-01-07_14-46.txt](https://github.com/oliexdev/openScale/files/5783659/openScale_2021-01-07_14-46.txt)

oliexdev, 2021-01-08:
I am confused which isn't showing correctly the lbs? What is user end? Which entry of log? Maybe some screenshots and photos will help to explain what you mean.

03172008, 2021-01-08:
By User End I meant in the app.

03172008, 2021-01-08:
it records the weight wrong, instead of saying, for example, 30 lbs, it shows 13 lbs, which is the kilogram measurement. because the weight is logged incorrectly, everything is offset, like the bmi.

mawcuzz, 2025-04-19:
I'm having this issue as well. It's amazing that I can even get data from a 7 year old weight scale is amazing. The lbs shows as the KG equivalent when the scale measures in lbs and I have the app set for lbs. The conversions then are off too due to this lb measurement.