oliexdev/openScale #681: Add support for SBF77

PeterHuewe, 2021-01-07:
**Scale name**
Silvercrest SBF77 https://www.lidl.de/de/silvercrest-diagnosewaage-sbf-77-mit-app/p361928 

**Step 1: Read the general reverse engineer process**

- [x] I have read the [how to reverse engineer a Bluetooth 4.x wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)  entry

**Step 2: Acquiring some Bluetooth traffic**
_Attach 3 log files with the corresponding true values, read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic) for further information._

1. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
male, 163, 35, activity level 3/5
```
measured true values in the vendors app for the 1. HCI Snoop log file:
```
weight, date, water, muscle , fat, bones
80,5 2021-01-07-20-38 48,1 37,6 29,1 2,9
```
[btsnoop_hci_20210107_201849.log](https://github.com/oliexdev/openScale/files/5784214/btsnoop_hci_20210107_201849.log)
BT Packets: 1828 - 1987

2. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
male, 163, 35, activity level 3/5
```
measured true values in the vendors app for the 2. HCI Snoop log file:
```
weight, date, water, muscle , fat, bones
82,0 2021-01-07-20-39 47,4 37,2 30,1 2,9
```
Same log, Pakets:1988-3105, Scale was NOT turned off
_--> Attach the 2. HCI Snoop log file here <--_

3. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
male, 163, 35, activity level 3/5
```
measured true values in the vendors app for the 3. HCI Snoop log file:
```
82,4 2021-01-07-20-41 47,2 37,1 30,4 2,9
```
Same log as above BT Packets: 3106-end

**Step 3: Discover Bluetooth services and characteristic**
_Read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#2-find-out-the-bluetooth-services-and-characteristic) how to create the openScale debug file_

[openScale_2021-01-07_22-40.txt](https://github.com/oliexdev/openScale/files/5784229/openScale_2021-01-07_22-40.txt)

**Reverse Engineering Progress**
Handle `0x3d` contains the timestamp of the measurement:
```
0edc3ee5070107142616022e015e06
xxxxxxYYYYMMDDHHmmxxxxxxxxxxxx
```

Handle `0x43`  should contain the weight measurements.

Handle `0x4A` contains the list of user(names) (`PH`)
```
0002 50 48 20    c1070a19 a3     0003
xxxx P  H  blank YYYYMMDD Height xxxx
```

I have Android Studio up and running and am willing to help to play around, if a suitable 'template' is available. The scale is different from SBF70/75, I already tried adding it as a compatible :)

PeterHuewe, 2021-01-07:
Seems to be related from the protocol to the SBF76 #550 
The scale is from Hans Dinklage GmbH and also linked to Beurer (https://www.beurer.com/web/we-dokumente/eukonformitaetserklaerungen/SBF_77__EC_DoC_Hans_Dinslage_20200724.pdf)

PeterHuewe, 2021-01-07:
Okay, interestingly, I just pulled a new hci log from my phone and wireshark is happily dissecting the protocol.
[image]
The weight seems to be doubled for some reason, but okay.
So it seems to talk 'standard' GATT bluetooth for body scales - see https://www.bluetooth.com/de/specifications/gatt/

Can somebody help me to create support for this scale? (which other scale is the best starting point)

PeterHuewe, 2021-01-08:
Okay, basically `BluetoothStandardWeightProfile.java` works with my scale :) - 
 I just had to uncomment the section in the Factory.java (why was the support for these standard scales commented out? @oliexdev )

A few issues remain though:
I had to hard code my user id (2) since the registration of a new user did not work (not authorized).
So I removed the user registration, and since I could not change the user id of my user in the app, as a workaround I just hardcoded it.

The consent code for my user/scale is different, I had to change it to 0x2688

Then I get readings from the scale, which look correct.
However, I get **two** entries in my database per weighing - one for the weight, the other for bodyfat & co (without the weight) :/ This is a bit unfortunate (see screenshot)
[image]

oliexdev, 2021-01-08:
> Okay, basically BluetoothStandardWeightProfile.java works with my scale :) -
> I just had to uncomment the section in the Factory.java (why was the support for these standard scales commented out? @oliexdev )

The reason is I never tested the standard weight profile code because I don't own that kind of scale and two things have to be done (as you already noticed) before it can be used. I already mention it at https://github.com/oliexdev/openScale/issues/502#issuecomment-620442379 (seems @felixandersen gave up).

> Then I get readings from the scale, which look correct.
> However, I get two entries in my database per weighing - one for the weight, the other for bodyfat & co (without the weight) :/ This is a bit unfortunate (see screenshot)

this should be surely fixed with some kind of logic.

PeterHuewe, 2021-01-08:
@oliexdev the scale is currently for 20 bucks or less at lidl - would you be up for getting one sponsored and make it fully supported in openscale? I think adding full support for standard bt scales would be a huge benefit for openscale. 

What would be your minimum requirements to have the support in the official pro version enabled? (even if not fully supported) (probably the pin entry?) 

Not yet sure where/how to add the pin entry dialog? Part of User Profile? (where one could also specify the user id) 

Not that familiar with the codebase yet :)

oliexdev, 2021-01-08:
> @oliexdev the scale is currently for 20 bucks or less at lidl - would you be up for getting one sponsored and make it fully supported in openscale? I think adding full support for standard bt scales would be a huge benefit for openscale.

Sure it would be great to support the standard BLE protocol but I currently don't have much time.

> What would be your minimum requirements to have the support in the official pro version enabled? (even if not fully supported) (probably the pin entry?)

That it works and easy to use.

> Not yet sure where/how to add the pin entry dialog? Part of User Profile? (where one could also specify the user id)

Since it is a specific scale problem and not a general one. The entry dialog should be shown during the first connection to the scale. So the pin entry dialog code needs to be implemented in the `BluetoothStandardWeightProfile.java`. The code itself could be saved to the local intern SharedPreferences.

> Not that familiar with the codebase yet :)

Well openScale is open-source ;) if you have specific question you can ask. I am willing to help you.

rapht0r, 2021-03-24:
@PeterHuewe Could you provide your modified files? I'd love to test this scale and try to support it. But I'm not sure where to find the Factory.java file you referred to.
Thanks in advance

krisjans, 2021-07-26:
@PeterHuewe I'm currently working to support Beurer scales that use Bluetooth Standard Weight Profile.
I suspect that SBF77 might be compatible with BF600 and BF850.
If you still have SBF77 scale and would like to test this theory - please let me know and I will create test apk for you.

PeterHuewe, 2021-07-26:
@krisjans yes I still have it and can test it.

rapht0r, 2021-07-26:
> @PeterHuewe I'm currently working to support Beurer scales that use Bluetooth Standard Weight Profile.
> I suspect that SBF77 might be compatible with BF600 and BF850.
> If you still have SBF77 scale and would like to test this theory - please let me know and I will create test apk for you.

I'd love to help test it, too

krisjans, 2021-07-26:
@PeterHuewe Thanks! Try my debug build: [openScale-2.3.5-debug_SBF77.zip](https://github.com/oliexdev/openScale/files/6877749/openScale-2.3.5-debug_SBF77.zip)

krisjans, 2021-07-26:
@PeterHuewe and @rapht0r I made a new class to support SBF77 (see changes in my [branch](https://github.com/krisjans/openScale/commits/SBF77)).
Here is debug build of my branch: [openScale-2.3.5-debug_SBF77_v2.zip](https://github.com/oliexdev/openScale/files/6879270/openScale-2.3.5-debug_SBF77_v2.zip).
Scale user selection and user consent/pin entry might work.
New user creation most likely will not work - I have no information how to set "user activity level". To discover those magic numbers someone with SBF77 must snoop Bluetooth while creating new users on scale with manufacturer's app.
Also I have no information how to initiate manual measurement from app - more logs needed.

If you manage to test it, please provide openScale debug logs (and optionally Bluetooth logs).

p.s. Thanks to @jensMF for writing user-list decoding function for Beurer BF105 - it looks the same for this scale too!

krisjans, 2021-07-28:
If someone is interested in testing my latest changes related to [BF600](https://github.com/oliexdev/openScale/issues/219), [BF850](https://github.com/oliexdev/openScale/issues/531), [BF950](https://github.com/oliexdev/openScale/issues/502) and [SBF77](https://github.com/oliexdev/openScale/issues/681), then latest automatic debug build is availabe here: [openScale-2.3.5-debug.apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk), change log in #753 and my [SBF77 branch](https://github.com/krisjans/openScale/commits/SBF77).

If something does not work or you encounter a crash - I would highly appreciate if you leave me a comment with attached openScale debug log (if possible also  btsnoop_hci.log. [Instructions how to get logs.](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic))

krisjans, 2021-08-10:
> If someone is interested in testing my latest changes related to [BF600](https://github.com/oliexdev/openScale/issues/219), [BF850](https://github.com/oliexdev/openScale/issues/531), [...]

Hello @PeterHuewe and @rapht0r !
I recently fixed issues related to BF950 (this one seems to be very similar to SBF77) and I'm pretty convinced that SBF77 should work now.
If you would like to participate in testing effor and you still have SBF77 available, then please test it with my  latest debug [apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk)

yoyossef, 2021-08-10:
I tested the debug apk on my SBF77 and it seems to work fine, it even created a new user on the scale, though the name on the scale shows "POS" instead of my initials.

krisjans, 2021-08-10:
@yoyossef  Thanks for the feedback!
> I tested the debug apk on my SBF77 and it seems to work fine, it even created a new user on the scale, though the name on the scale shows "POS" instead of my initials.

Would you like to enable debug logs in openScale, test it again (create new user and do few weight measurements) and post the log here?

yoyossef, 2021-08-10:
@krisjans Sure:  [openScale_2021-08-10_14-10.txt](https://github.com/oliexdev/openScale/files/6961576/openScale_2021-08-10_14-10.txt)

krisjans, 2021-08-24:
Anyone willing to help out and test latest changes for #753 on SBF77?
Latest automatic debug build is availabe here: [openScale-2.3.5-debug.apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk).
I'm especially interested if user initials is now stored and displayed on SBF77 when you create new user from openScale.

p.s. @yoyossef maybe you would like to help me once more and test this build and provide your comments and debug log?

krisjans, 2021-09-10:
> Anyone willing to help out and test latest changes for #753 on SBF77?
> Latest automatic debug build is availabe here: [...]

I just made a refactoring that might affect SBG77. 
I would highly appreciate if anyone could test my latest debug [apk](https://github.com/krisjans/openScale/releases/download/debug_latest/openScale-2.3.5-debug.apk) and let me know if it still works on SBF77.

yoyossef, 2021-09-14:
@krisjans I can confirm that it still works on the SBF77.