oliexdev/openScale #683: wger (open source workout manager) weight sync support

gudvinr, 2021-01-10:
**Is your feature request related to a problem? Please describe.**
There aren't many open source health trackers but probably most known (or only existing) is [wger](https://github.com/wger-project/wger) which has [REST API](https://wger.de/en/software/api) too.

**Describe the solution you'd like**
It would be useful to have wger support to sync along with google fit and mqtt.

**Describe alternatives you've considered**
Possible alternative is to support openscale MQTT protocol. But openScale is not a service from where you can "pull" the data but rather "push"-based client.  
MQTT output which exists in sync is only useful for services you write yourself otherwise it isn't good practise to support multiple client APIs in single service.  

Also there's no "standard" API specifications for this kind of thing so you already must implement healthkit/google fit integrations different.

**Additional context**

[Here](https://wger.de/en/software/api) is the documentation for API.  
You must know beforehand server endpoint (because, well, it is self-hosted) but you may use `https://wger.de/api/v2/` as default one.

Before sending requests you must send `POST /api/v2/login` to obtain token using `Content-Type: application/json` with payload containing `username` and `password`. It is a bit unclear from documentation so I made an [issue](https://github.com/wger-project/wger/issues/598) here.

After that you may send `POST api/v2/weightentry/` using `Authorization: Token` and `Content-Type: application/json` with `date` and `weight` fields.

oliexdev, 2021-01-11:
I actually already played a bit around with the wger API, see https://github.com/wger-project/wger/issues/448
But currently the DELETE and PATCH operation is not allowed, which prevent it to sync with openScale.

gudvinr, 2021-01-11:
Thanks, good to know

rolandgeider, 2021-01-13:
FYI that was only for some of the exercise related endpoints (we recently changed some of the DB structure and still need to make sure the submission process works), I should have been clearer there. The weight endpoint should have no problems with PATCH and DELETE (and if it does, just ping me)

oliexdev, 2021-01-14:
@rolandgeider unfortunately, for me it still not work at https://wger.de/api/v2/weightentry/ there is no PATCH or DELETE operation allowed:

> Allow:
>                     GET, POST, HEAD, OPTIONS

But I might do something wrong? If I try to delete something, I always get this back as a response `{"detail":"Method \"DELETE\" not allowed."}`

rolandgeider, 2021-01-14:
That is strange. Did you send the auth header? I'm developing a flutter app that talks to the server via the regular REST API and it definitely works

oliexdev, 2021-01-15:
I used this command (XYZ is replaced with the correct token)

> curl -X DELETE https://wger.de/api/v2/weightentry -H 'Authorization: Token XYZ' -d '{"date":"2020-12-04","weight":"77"}' -H "Content-type: application/json"

rolandgeider, 2021-01-17:
It does look like a bug, that should work

rolandgeider, 2021-01-22:
I have found what's wrong, hadn't seen it in the beginning. To PATCH a specific weight entry you'd use `/v2/weightentry/<id>/`, while '/v2/weightentry/' is for overviews and POSTing new ones.

oliexdev, 2021-02-03:
@rolandgeider thanks it works now :+1: sorry for the confusion

oliexdev, 2021-02-19:
@rolandgeider I have implemented a first version for synchronization openScale sync with wger. It works good but I have two remarks:
1. In the wger REST API you get/save the date in the yyyy-MM-dd format, which prevent you to add multiply weights for one day. I would suggest to get/save the date in msec or millisec (regarding to the Unix Epoch Time). 
2. You use in the REST API an id to identify the weight entry. Here it would be again better to use the date in msec/millisec for the identifier. Currently I need to fetch the whole wger weight entry list to search for the wger id to update or delete a wger weight entry.

Would you consider to change that in your REST API? If yes how fast could that be implemented in wger?

rolandgeider, 2021-02-21:
Nice!!

Yes, there is also a uniqueness constraint in the model, so that only one entry per user and date can be saved. Adding more wouldn't make much sense for an app like wger and makes things like plotting the chart easier if there aren't duplicate entries.

And as for the ID, you can search if there are entries for a specific date with ``/api/v2/weightentry/?date=2021-02-21``

oliexdev, 2021-02-21:
@rolandgeider thanks for the hint with the date query, I use this instead of fetching all data. I have been released the openScale sync 0.3.2 , which should support the synchronization with wger. Thanks for your open-source workout manager. :medal_sports:

closed 2021-02-21 by oliexdev