oliexdev/openScale #686: Runtastic Libra - NullPointerException: Attempt to read from field 'int

Janfraaank, 2021-01-14:
I'm having trouble when using OpenScale with Runtastic Libra scale. 
No metrics are transmitted to the app. 

I see is an exception in the log (attached). 

Thank you for your work.

BluetoothBeurerSanitas: java.lang.NullPointerException: Attempt to read from field 'int com.health.openscale.core.bluetooth.BluetoothBeurerSanitas$RemoteUser.localUserId' on a null object reference
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.processMeasurementData(BluetoothBeurerSanitas.java:350)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.processMeasurement(BluetoothBeurerSanitas.java:401)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.onBluetoothNotify(BluetoothBeurerSanitas.java:282)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate(BluetoothCommunication.java:358)
	at com.welie.blessed.BluetoothPeripheral$1$5.run(BluetoothPeripheral.java:401)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:237)
	at android.app.ActivityThread.main(ActivityThread.java:7948)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1075)
[openScale_2021-01-14_17-19.txt](https://github.com/oliexdev/openScale/files/5815826/openScale_2021-01-14_17-19.txt)

oliexdev, 2021-01-15:
see #552

closed 2021-01-15 by oliexdev

Janfraaank, 2021-01-15:
Got it and resolved!
Sorry for duplicate