oliexdev/openScale #688: Allow weight measurement in grams

bohallaron, 2021-01-18:
**Is your feature request related to a problem? Please describe.**

I have been using this app for a few months to track the weight of my pet bunnies, who weigh around 1kg each. Recording weights in kg is not intuitive, and anyway the weight always gets rounded by the app to 2 decimal places.

**Describe the solution you'd like**

Simply add the option to measure weight in grams to expand the current options of kg/stone/lb.

**Describe alternatives you've considered**

I really only need the functionality to track weight and chart using a line graph, but this simple functionality isn't so easy to find in other apps. My other alternative would be making an excel spreadsheet to do this.

oliexdev, 2021-01-18:
the lowest possible weight is 10g (0,01kg) which is sufficient for most of the use cases. I don't see a problem to enter grams into openScale.
It has to be rounded to two decimal places because if we convert for example kg->lb there are always more decimal places which are not useful anymore.

closed 2021-01-18 by oliexdev

bohallaron, 2021-01-18:
That's OK if you don't see a problem, but as a user it makes the app less usable for me than it could be and is something that would expand the use of the app (there are lots of people with pets) and seemed like a very simple addition. What is confusing is that 3 decimal places are displayed in the graphs, but the third one is apparently just clutter since you require that it rounds to 2 anyway.