oliexdev/openScale #689: Problems with a new Renpho Scale (ES-CS20M)

linwin1, 2021-01-19:
I cannot connect to the above scale. It isn't even found but the RenPho app works as expected.
Maybe someone has some informatin about this scale. I have searched already but it looks like there are different versions of this scale. The wireshark log is attached.

I hope someone can give an advise.

[bluetoothd-hci-latest.zip](https://github.com/oliexdev/openScale/files/5834412/bluetoothd-hci-latest.zip)

oliexdev, 2021-01-19:
please read the FAQ

closed 2021-01-19 by oliexdev

linwin1, 2021-01-19:
I have read the FAQ and I have tested the developement version without success.

oliexdev, 2021-01-19:
you should read this https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-(FAQ)#why-is-my-scale-not-found

linwin1, 2021-01-19:
I have read this FAQ. The scale offers an advertisment but is not connectable. Pairing doesn't work for this device. Please have a look at the file attached in my first question.

linwin1, 2021-01-19:
Here is the btsnoop_hci.log recorded by my Sony tablet. The relevant part starts at a time offset of 38 secondes. To my opinion there is no connectin between the devices. I think the data of the scale is send in the advertisment message. 
[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/5836575/btsnoop_hci.log)