oliexdev/openScale #693: First measurement after opening app has wrong date and time

ThomasRadeke, 2021-01-21:
**Describe the bug**
Upon first measuring after opening openScale on my Android, the measurement has the wrong date and time - the date is usually from the previous measurement, but I have no idea where the time is coming from.
If I keep the app open, let the scale reset and retake the measurement, all values are correct.
I'm using a Medisana BS444.

**To Reproduce**
Open openScale (BT Autoconnect is enabled) and step on the scale.
I'm using the F-Droid version, which says v2.3.5 (54).

**Expected behavior**
Correct time and date.

**Additional context**
In Medisana's own app, VitaDock+, it works every time, so it's probably not the scale. :-)

**Debug log**
[openScale_2021-01-21_15-09.txt](https://github.com/oliexdev/openScale/files/5849834/openScale_2021-01-21_15-09.txt)

oliexdev, 2021-02-03:
and what are the correct date and time?

ThomasRadeke, 2021-02-03:
The log time from line 39 would be the correct one: "2021-01-21 15:09:48.236".

On line 87 of the log, you can see a correct measurement - log time is "2021-01-21 15:11:06.642" and the "DATE_TIME" field of the measurement is "Thu Jan 21 15:11:03 GMT+01:00 2021". The seconds are a bit off, but that's not a big deal, I guess. At least it got the date, hours and minutes right. :D

oliexdev, 2021-02-03:
is this always the case? Could be a timing problem because the date/time is set in step 3.

ThomasRadeke, 2021-02-03:
Yes, it has always been like this since I started using openScale. I only have this physical scale to try but I'm open for suggestions.

oliexdev, 2021-02-03:
but you doing to sperate measurements? One has the correct date/time and an incorrect date/time?

It's a bit strange maybe you could play a bit around in the [medisana source-code](https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothMedisanaBS44x.java)

ThomasRadeke, 2021-02-03:
That's difficult for me because I have no experience whatsoever in developing Android apps. I'll see what I can do.

vonneudeck, 2021-02-06:
I have the same scale and the same issue. Once in a while something changes though and then I get the correct time and date. Sometimes for a few days. Then I get the faulty behaviour.

tukusejssirs, 2021-10-02:
> The only constant is that I always get the message about using the scale barefoot. Even though I am using it barefoot.

Maybe I should open a separate issue, but
- I don’t have the original issue (incorrect timestamp); all my measements have a correct timestamp;
- I measures only the following data (both in openScale and VitaDock+):
   - BMI;
   - BMR;
   - TDEE;
- I also always get the toast with about barefoot measurement (which I do).

Note that I’ve just bougth the scale, haven’t done any configuration on it (should do anything?).

tukusejssirs, 2021-10-02:
> Maybe I should open a separate issue, but
>
> * I don’t have the original issue (incorrect timestamp); all my measements have a correct timestamp;
> * I measures only the following data (both in [...]

closed 2023-02-12 by oliexdev

stefanct, 2025-04-22:
FWIW, I just connected to my new BS 444 (that was never paired to any other device) and the measurement date was also incorrect (I only remember the year was 2031). Subsequent measurements seem to be fine... so it's not a big issue at all but something's wrong in the app. I guess it can't be uninitialized variables directly but maybe there is a quirk where uninitialized kernel data is received or something?