oliexdev/openScale #694: support for iOS?

Richter-Ko, 2021-01-22:
is it possible to work it on iOS?

oliexdev, 2021-02-03:
no because it's an Android app

closed 2021-02-03 by oliexdev

akirataguchi115, 2022-11-28:
I just found this issue while searching for iOS support. Any possibility to re-open this issue?

I am fustrated how the iOS Health Coach app has stopped communicating my scale data to my iPhone.

I would want a iOS version of this app because @MartinNowack seems to have implemented support for SBF 70 which I have.

I've considered just getting rid of the scale due to the aforementioned lack of support on iOS.

oliexdev, 2022-12-09:
sorry to support iOS you have to rewrite the whole app which I definitely not have the time for.

akirataguchi115, 2022-12-09:
Maybe if this issue was open someone could rewrite it in some framework that is multiplaform?