oliexdev/openScale #696: Am I missing something? Can't connect Renpho ES-CS20M

DuckeeUK, 2021-01-31:
I've just bought a Renpho ES-CS20M Scale since it appeared to be supported, but I'm not able to get it to connect.  Am I missing something or doing something wrong here?

STEPS I'M TAKING TO TRY TO CONNECT
1. Bluetooth is OFF, Location permission is GRANTED and Location is ON.
2. Open the app
3. Stand on the scale to wake it
2. In app search for device, granting permission to use BT.
3. It fails to find the scale

I've also tried repeating the same steps but with BT already on on my phone in step 1, still no device found.

I read the FAQ and I've tried 3 different sets of brand new batteries - no joy!

My phone (Sony Xperia 10 II, Android 10) also would not find the scale when I tested pairing via the phone's BT search.  Does this indicate it's a faulty scale or it's not compatible?

oliexdev, 2021-02-03:
Try out the vendors app, if it still not work then you have faulty scale, the rest is written in the FAQ.

closed 2021-02-03 by oliexdev

DuckeeUK, 2021-02-03:
Like I said, I read the FAQ, I did as the FAQ suggested.  Thanks for your "support".

oliexdev, 2021-02-03:
your welcome using the open source app and taking my free time.

DuckeeUK, 2021-02-03:
I bought a scale on the basis that you have it LISTED AS COMPATIBLE when clearly there is an issue with it.  Generally I am super appreciative of devs who make open source software for free, but when I have PAID MONEY for a scale based on your listing it as compatible, then yes, I'm more than peeved at your attitude of "just look at the FAQ - ISSUE CLOSED" when I've ALREADY done that!   So excuse me for not being super appreciative for an app that's not even working for me.

Goodbye.

oliexdev, 2021-02-03:
Feel free to fix it and send a PR. I don't even own this scale and I even write a FAQ about this problem.

If you don't even find your scale via Bluetooth it has NOTHING to do with the app!

I feeling really motivated to develop or maintain openScale with just an unthankful attitude.

matrottier, 2021-02-03:
@DuckeeUK  is it the model ES-CS20M-W ?  
the vendor app working for me but sadly not openscale (dev).

guix77, 2021-09-02:
See #769

mikewilzn, 2022-02-02:
> Feel free to fix it and send a PR. I don't even own this scale and I even write a FAQ about this problem.
> 
> If you don't even find your scale via Bluetooth it has NOTHING to do with the app!
> 
> I feeling really motivated to develop or maintain openScale with just an unthankful attitude.

We appreciate the work you've done. It's a great app. We're simply asking that it be removed from the wiki as a "supported scale" considering it is obviously not a supported scale.