oliexdev/openScale #697: No bluetooth debug information in log

APIUM, 2021-02-08:
**Describe the bug**
No bluetooth devices found in app or log. Using a Renpho ES-CS20M. Phone is a Samsung S20+.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) #c64809a

**Expected behavior**
I would expect the scale to have appeared in the app as there is support for it, if not that I would see an unrecognized device in the logfile.

**Debug log**
```
2021-02-08 13:03:38.104 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v2.3.2-dev_5d697fe9_2020-11-05 (1604584993), SDK 30, samsung SM-G985F
2021-02-08 13:03:38.107 Debug [2] AboutPreferences: Selected user id(1) name(test) birthday(Thu Feb 08 11:42:14 GMT+11:00 2001) age(20) body height(184.00) scale unit(kg) gender(male) initial weight(82.00) goal weight(80.00) goal date(Sun Aug 08 11:42:14 GMT+10:00 2021) measure unt(cm) activity level(2) assisted weighing(false)
2021-02-08 13:04:01.879 Info [2] BluetoothCentral: scan started
2021-02-08 13:04:21.886 Info [2] BluetoothCentral: scan stopped
2021-02-08 13:04:24.271 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-02-08 13:04:26.015 Info [2] BluetoothCentral: scan started
2021-02-08 13:04:46.021 Info [2] BluetoothCentral: scan stopped
2021-02-08 13:04:50.033 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-02-08 13:04:57.180 Info [2] BluetoothCentral: scan started
2021-02-08 13:05:05.009 Info [2] BluetoothCentral: scan stopped
2021-02-08 13:05:05.150 Info [2] BluetoothCentral: scan started
2021-02-08 13:05:07.652 Info [2] BluetoothCentral: scan stopped
2021-02-08 13:05:07.726 Info [2] BluetoothCentral: scan started
2021-02-08 13:05:27.730 Info [2] BluetoothCentral: scan stopped
2021-02-08 13:05:28.330 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-02-08 13:05:46.801 Debug [2] AboutPreferences: Debug log disable
```

Should I look at getting a dump of the bluetooth traffic and giving it an analyse? Can someone please point me towards the file that implements support for the RENPHO? I couldn't find it in https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/ which is where I expected it.

oliexdev, 2021-02-17:
please read the FAQ

closed 2021-02-17 by oliexdev

guix77, 2021-09-02:
See #769