oliexdev/openScale #699: Ugly UI

mammuthus, 2021-02-13:
Hi, guys. Why UI is so ugly? 
Any plans to make views more beautiful and attractive?
How can the community help?

dror3go, 2021-02-16:
Oh wow, guess you can start by giving constructive criticism instead of that useless comment.
How about offering UX/UI or by providing PR with some relevant changes?

mammuthus, 2021-02-16:
>How about offering UX/UI

Nice idea, thank you! We just have to figure out if the authors are interested.

PavelGorbanj, 2021-02-17:
UI is quite straightforward/self-explanatory and simplistic

mammuthus, 2021-02-17:
@PavelGorbanj Yeah, I totally agree. Unfortunately, it doesn’t stop it from being ugly and inconsistent.
Oh, these alignments and font styles, #000 header... yuk. Proximity law, Fitts's? No way.

PavelGorbanj, 2021-02-17:
> @PavelGorbanj Yeah, I totally agree. Unfortunately, it doesn’t stop it from being ugly and inconsistent.
> Oh, these alignments and font styles, #000 header... yuk. Proximity law, Fitts's? No way.
beauty is in the eye of the beholder; ugliness is "as well", I guess

It looks like it should be quite easy for you to draft high-quality (from both UX and UI point of view) mocks/flows that will preserve simplicity of the solution, it's flexibility and account for optimal efforts invested into rework, assuming it's kinda free app.
If result looks good, it may inspire others to help with rework.

mammuthus, 2021-02-17:
@PavelGorbanj Yeah, hive mind is an advantage of OS-projects.
So we just have to figure out if the authors are interested. Sure you will attend too, as a reviewer at least.

oliexdev, 2021-02-17:
> Nice idea, thank you! We just have to figure out if the authors are interested.

@mammuthus Go ahead, I am already exciting in your state of the art UI/UX design proposals. I am waiting of your pull request.

oliexdev, 2021-03-15:
As I thought, it was nothing but hot air. Anyone can badmouth the app and talk catchy phrases, but when it comes to the implementation then @mammuthus is still. How pathetic.

closed 2021-03-15 by oliexdev

mammuthus, 2021-03-15:
Don't remember what you set conditions or a deadline for me. Some of mockups (overvies/stats screens) were already published on figma.com/community to obtain the input, but okay, as the owner your opinion is valuable for me.
`So we just have to figure out if the authors are interested. `
Not interested. Duly noted, glory luck :)

oliexdev, 2021-03-15:
nothing had been published and posted here. Empty words.