oliexdev/openScale #701: Crash when opening openscale sync

flombleblock, 2021-02-16:
**Describe the bug**
I opened the Openscale Sync app and got a message that it crashed and to report it.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the the phone's apps screen
2. Click on openscale Sync
3. See error

**Expected behavior**
Expected the application's Overview page

**Debug log**

Build version: 0.3.1 
Build date: 1979-11-30 00:00:00 
Current date: 2021-02-16 09:42:08 
Device: Samsung SM-G975U 
 
Stack trace:  
android.app.RemoteServiceException: Context.startForegroundService() did not then call Service.startForeground(): ServiceRecord{caf6e71 u0 com.health.openscale.sync/.core.service.SyncService}
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2210)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:246)
	at android.app.ActivityThread.main(ActivityThread.java:8425)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:596)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)

oliexdev, 2021-02-17:
Strange do you get it every time on the startup? Could you reinstall the app? Which Android OS version do you have on your smartphone?

flombleblock, 2021-02-17:
Just happened one time. Worked before and when I hit the restart button it worked again. Phone is running Android 11 (probably upgraded a month or so back)

oliexdev, 2021-02-17:
ok then I will close this issue, since it happened one time.

closed 2021-02-17 by oliexdev