oliexdev/openScale #703: Crash on setup with XiaoMi scale v2

deeeed, 2021-02-24:
**Describe the bug**
During first launch and setup of the application, application crash during scale setup.
Fail to import history. Error happens with latest paid pro version and latest dev version.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): yes

**Expected behavior**
It should import existing history from xiaomi scale.

**Additional context**
[image]

**Debug log**
[21:15, 2/24/2021] Arthur: Build version: 2.3.5-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2021-02-24 21:14:19 
Device: Samsung SM-G9750 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{980fc78} (9c93fca3-1622-4c83-b5c0-4eaadf103325)} not attached to a context.
	at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:205)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1$1.run(BluetoothSettingsFragment.java:132)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:246)
	at android.app.ActivityThread.main(ActivityThread.java:8425)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:596)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
[21:49, 2/24/2021] Arthur: Build version: 2.3.2-dev_5d697fe9_2020-11-05 
Build date: 1981-01-01 01:01:02 
Current date: 2021-02-24 21:49:24 
Device: Samsung SM-G9750 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.IllegalArgumentException: ID does not reference a View inside this Activity
	at android.app.Activity.requireViewById(Activity.java:3430)
	at androidx.core.app.ActivityCompat.requireViewById(ActivityCompat.java:363)
	at androidx.navigation.Navigation.findNavController(Navigation.java:58)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$BluetoothDeviceView.onClick(BluetoothSettingsFragment.java:365)
	at android.view.View.performClick(View.java:8160)
	at android.widget.TextView.performClick(TextView.java:16104)
	at android.view.View.performClickInternal(View.java:8137)
	at android.view.View.access$3700(View.java:888)
	at android.view.View$PerformClick.run(View.java:30250)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:246)
	at android.app.ActivityThread.main(ActivityThread.java:8425)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:596)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)

oliexdev, 2021-03-04:
does this crash happens everytime?

KyleHaynes, 2021-03-11:
Current version appears to crash for me with same brand scales.

Regarding the question ...

> does this crash happens everytime?

It occurs as soon as the sync takes place (so if the setting option is to sync automatically on startup, it will crash everytime you start the app.

> Build version: 2.3.2-dev_5d697fe9_2020-11-05 
Build date: 1981-01-01 01:01:02 
Current date: 2021-03-12 04:56:29 
Device: Samsung SM-A205YN 
OS version: Android 10 (SDK 29) 
 
> Stack trace:  
java.lang.ArrayIndexOutOfBoundsException: length=4; index=4
	at com.health.openscale.core.utils.PolynomialFitter.gj_swap(PolynomialFitter.java:186)
	at com.health.openscale.core.utils.PolynomialFitter.gj_echelonize(PolynomialFitter.java:156)
	at com.health.openscale.core.utils.PolynomialFitter.getBestFit(PolynomialFitter.java:118)
	at com.health.openscale.gui.measurement.ChartMeasurementView.addPredictionLine(ChartMeasurementView.java:535)
	at com.health.openscale.gui.measurement.ChartMeasurementView.addTrendLine(ChartMeasurementView.java:512)
	at com.health.openscale.gui.measurement.ChartMeasurementView.refreshMeasurementList(ChartMeasurementView.java:392)
	at com.health.openscale.gui.measurement.ChartMeasurementView.updateMeasurementList(ChartMeasurementView.java:347)
	at com.health.openscale.gui.graph.GraphFragment$10.onChanged(GraphFragment.java:283)
	at com.health.openscale.gui.graph.GraphFragment$10.onChanged(GraphFragment.java:280)
	at androidx.lifecycle.LiveData.considerNotify(LiveData.java:131)
	at androidx.lifecycle.LiveData.dispatchingValue(LiveData.java:149)
	at androidx.lifecycle.LiveData.setValue(LiveData.java:307)
	at androidx.lifecycle.LiveData$1.run(LiveData.java:91)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:237)
	at android.app.ActivityThread.main(ActivityThread.java:7948)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1075)

oliexdev, 2021-03-15:
@KyleHaynes you have a different bug bug this bug is already fixed please update to the latest openScale version.

closed 2021-03-15 by oliexdev