oliexdev/openScale #716: Make Week & Month (and year) average come back (Please)

SuzukaDev, 2021-03-22:
Older versions of OpenScale had the option to **see the average for each week (and month (and year))**. It was removed at some point the last year, IIRC, and now is **missing**. I waited for a few updates, and I thought it was "deprecated" by error and it would be added again, since is such a trivial and useful feature. But it didn't happen.

Another user opened another similiar issue:
https://github.com/oliexdev/openScale/issues/670

It was told to "just change the view in the graph to week". But, no. Changing the graph view to week (or month), only changes the labels on the Y-axis to week (or months).
But **the graph is exactly the same, there is no week/month/year average**.

Could it be added again, please?
I'm not asking for a new feauture, just to incorporate again that missing functionality that was removed.
Personally, it was my favourite feature, and the thing that helped me the most to see and track my progress. Daily measures are not as precise as week averages to track the progress, and since it was removed, I'm making the average via pen&paper... wich is weird, because thats why I use(d) the app, to track those things for me.

Thank you so much!

john-bill, 2021-04-17:
don't know when it as been removed, but its really missing.
But I'm also missing the (very-)old Statistics-View, when we had statistics instead of useless (and from day one wrong!) diagrams...

oliexdev, 2023-07-23:
Thanks @undefiened we have a new option which use a simple moving average on the chart.

closed 2023-07-23 by oliexdev