oliexdev/openScale #717: SBF70 connection issues

holgerpieta, 2021-03-27:
**Describe the bug**
I'm aware that very similar bugs have been submitted before and that the underlying problem is a timing problem in the scale itself. But I'm willing to dig into that and try to come up with a solution that makes things more stable, e.g. by retrying when timing failed. But I still like to have this bug open here as a place to discuss ideas and possible solutions.

Most important thing I noticed: After the first unhandled exception the app itself seems to end up in a undefined state where it will not be able to connect until it is hard-killed and restarted. So I guess that's the problem I need to solve first.

I will need to learn how to build an android app first, so it may take a couple of days before I have new ideas.

If anyone has good ideas about what to try first, please bring them up.

**To Reproduce**
Steps to reproduce the behavior:
1. Open app
2. Click Bluetooth button
3. Step on scale
4. Wait for scale to finish measuring
5. No new measurement visible in app
6. Repeat as often as you want
7. Kill app
8. Open app
9. Touch scale to activate, but do not step on it
10. Click Bluetooth button
11. Measurement is transmitted

Reproduced with 2.3.5-pro (54) on two devices.

**Expected behavior**
Measurement is transmitted.

**Debug log**
Here is an exemplary log file. Unfortunately the logging disabled itself between step 7 and 8, so I do not have a log of the successful connection. I'll add that later.
[openScale_2021-03-27_09-04.txt](https://github.com/oliexdev/openScale/files/6215548/openScale_2021-03-27_09-04.txt)

holgerpieta, 2021-03-28:
Found something: The scale, for some reason (bug, paranoia, whatever), sends the reply to `CMD_SCALE_STATUS` up to three times.
The app receives the first message and advances to the next step. But, as far as I understand the BLESSED library, the additional replies will still be in the queue and are thereby "received" in the next step.
Unfortunately, the state machine is not prepared for messages not matching the current state, receives the non-matching message and advances to the next step nonetheless.
The result is chaos.

It might be a good idea to empty the receive queue before a new command is issued, but I cannot find a way to do that in the BLESSED library. So I think we have to make the state machine more robust against non-matching replies.
I'll look into that.

leahoswald, 2021-04-22:
@oliexdev could you please consider merging the associated pull request? For me the SBF70 isn't usable for over a month now despite a lot of tires.

andrerodr, 2021-04-22:
My Sbf70 has been working fine all the time!

oliexdev, 2021-04-22:
please try out the latest development version

closed 2021-04-22 by oliexdev

holgerpieta, 2021-04-22:
Previously connection to the scale depended a lot on what you did in what order, e.g. one of:
- Start app, tap Bluetooth button, step on scale.
- Start app, tap Bluetooth button, tap scale to start, wait for connection, then step on scale.
- Step on scale, start app,  tap Bluetooth button.
- Tap scale to start, start app, tap Bluetooth button, wait for connection, step on scale.

Some procedures worked, some not. And that even depended on which Android device you used. For example my phone worked mostly (but not always) using procedure 1. My wife's phone only worked with procedure 4 and quite frequently not at all.

So I reworked the state machine to be much more robust and now all procedures "should" work.

On my phone it actually does work. But my wife's phone still doesn't like it quite frequently. But that's not a problem of the state machine, but the Bluetooth connection fails often with GATT_ERROR before the state machine even starts properly.
Rarely it works on first try, sometimes it works on the automatic retry, nearly always it works by manually retrying after it stopped after the second failed attempt.
I'm currently looking into the Bluetooth code to see how we can make it retry more often if it doesn't work the first time.

leahoswald, 2021-04-25:
Got the time to test it now and even if it doesn't work on the first try it worked on the second attempt. So clearly an improvement. (tested on an Pixel 5)

holgerpieta, 2021-04-25:
I opened a new issue #726 to discuss this problem.
@leahoswald: Please attach your debug log to the new issue, then I'll try to find a solution for our common problem.

schoenf, 2021-05-03:
1.9.3 is the last version that works well with my SANITAS SBF70. The latest version connects via Bluetooth, the smartphone says "connected" but the scale does not even highlight the "B"-Symbol, nor does it show the nickname nor does it transmit any measures. I'm keen on trying the latest development version to see if it fixes the problem.

holgerpieta, 2021-05-06:
This is exactly the kind of problems I tried to solve with my updates, so I'm mildly optimistic that the new version should make you happy.
@oliexdev released a development version 3 days ago: https://github.com/oliexdev/openScale/releases/tag/travis-dev-build
The description is somewhat confusing, because it is talking about 87775f6512eb0be1296e4a1ac44c228d0d894fe2 and 2020-01-21, but if you look at the left side, you see it's actually based on 0f5aad1ff8671c3bbff839218460eed8c40f25b6 three days ago. So it should contain all the fixes.

I had been chasing some BlueTooth issues lately, until, suddenly, the scale couldn't be found at all anymore. After resetting it by pulling the battery for a couple of seconds, everything worked fine and even the intermittent BlueTooth issues went away. So keep that in mind, too. Maybe the scale needs a hard reset every now and then.

PR #718 (2021-04-22): SBF70: Make BeurerSanitas state machine more robust
The SBF70 has three bad habits:
- It sends the scale status reply multiple times (three, as it seems).
- When the user is standing on the scale, measurement are sent out unrequested.
- In both situations the scale ignores other commands.

Previously, the BeurerSanitas state machine was not robust against those effects:
- When it received messages that didn't not match the current state, it sometimes still advanced to the next state. This caused missing information and undefined behavior in later steps.
- When it sent out a request while the s [... cut]