oliexdev/openScale #726: BlueTooth connection issues: GATT_ERROR (SBF70 and others)

holgerpieta, 2021-04-25:
Because #717 has been closed and this problem is different anyway, here is a new ticket to discuss this problem.

On some combinations of scale (e.g. SBF70) and phone (e.g. Sony Xperia XZ1 Compact) BlueTooth connection fails frequently on first try with GATT_ERROR. Sometimes the second (automatic) attempt succeeds, but frequently it doesn't.
Retrying manually works nearly every time.

So I guess we need to update the BlueTooth code to retry more often and, if possible, retry faster.

If anyone has a debug log of failed connections, please attach them here. Then I can see if it's the same problem as what I get.

hrkrx, 2021-07-08:
I have a SBF75 and the same error, see log:

[openScale_2021-07-08_22-26.txt](https://github.com/oliexdev/openScale/files/6787130/openScale_2021-07-08_22-26.txt)

I found it more reliable when you start scanning for the scale before you turn it on, and just step on it while searching.

holgerpieta, 2021-07-09:
Interesting, this GATT_ERROR happened a lot later in the state progression that it usually does for me.

Anyway, I have sort of stopped debugging this, because I found that when I get connection errors too frequently, I just have to restart the scale by taking out the batteries for a few seconds. After that connection is again reliable for months.
So looks like the scales software is the real problem and not exactly stable, going into some broken state after a while. This confused me for a while, because I though the scale would be off when the display is off and it would do a full restart when you step on it. But then I found that you can actually connect to the scale while it is "off", so I guess it never really shuts down except when you remove the battery.
So, like any other computer (doh!) it appears to need a restart every now and then.