oliexdev/openScale #727: Import from HealthMate

KrOnAsK, 2021-04-26:
Hi, I am trying to import my data from the Withings HealthMate app and it just doesn't seem to work, I already tried to edit the csv file and now I got the headers date,weight,fat,water,muscle,bone but every time I try and import the data to openscale it just tells me the formatting is wrong. Can anyone help me pls? or maybe someone already wrote a script which I didn't find.

Date | Weight | Fat | Water | Muscle | Bone
########## | 69,61451247 | 6,439909297 | 43,53741497 | 60,1814059 | 2,947845805

Thats a sample row of my csv file

beohna, 2021-04-30:
Hi, I am also new to the openscale-app and I use a SBF70. 
I made a export and the csv looks like this:
"biceps","bone","caliper1","caliper2","caliper3","calories","chest","comment","dateTime","fat","hip","lbm","muscle","neck","thigh","visceralFat","waist","water","weight"
0.0,12.85,0.0,0.0,0.0,0.0,0.0,"","2021-04-18 11:51",20.3,0.0,0.0,40.5,0.0,0.0,0.0,0.0,53.0,77.35

Maybe, when you transform your csv in that format, you can import it to openscale? Let me know if it works!

KrOnAsK, 2021-04-30:
could u maybe send me that empty csv file you made ?  Cause mine doesn't seem to work and maybe if I use yours and just enter my data it could work

oliexdev, 2021-05-16:
see the wikipage for import scripts references https://github.com/oliexdev/openScale/wiki/Useful-import-scripts
If you want to know how the openScale csv looks like, just export a sample measurement using openScale.

KrOnAsK, 2021-05-21:
Thanks, but I am not able to write scripts and was hoping someone else could do that for me. I can provide the csv file from the health mate app but I don't know how to transform that into the openScale csv format.

fds2610, 2024-01-02:
Has anybody succeeded in importing withings data automatically, yet?