oliexdev/openScale #728: Import vom HealthCoach xls

beohna, 2021-04-27:
Hi, 
I am trying to import my data from the SBF70 Breuer/Sanitas. The data can exported from the  HealthCoach Website (https://connect.sanitas-online.de/HealthCoach/Modules/Devices/DeviceOverview.aspx) as Excel Export.

I want to transform the xls-file manualy to csv-file, but i didn`t know the necessary data-format of the csv to import in openscale.

Here a example of the HealthCoach export s:

Date | Time | Weight kg | BF | BF, top % | BF, bottom % | Water | Muscle % | Muscle, top % | Muscle, bottom % | Bone kg | BMI | BMR kcal | AMR kcal | Activity level | Waist cm | Hip cm | WHR | Target weight kg | Manual | Analysis | Comment
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
19.12.2018 | 22:34 | 75,7 | 16,8 |   |   | 56,3 | 41,9 |   |   | 10,7 | 23,1 | 1674 | 2511 | 3 |   |   |   |   | No | Yes |  

A documentation of the import-format and export-format for openscale in the wiki would be a good idea!?

oliexdev, 2021-05-16:
see the wikipage for import scripts references https://github.com/oliexdev/openScale/wiki/Useful-import-scripts
If you want to know how the openScale csv looks like, just export a sample measurement using openScale.

sarika-03, 2025-05-23:
> [@sarika-03](https://github.com/sarika-03) Please post your script here

Hi! 
I’ve created a small Python script that converts the HealthCoach Excel export (from Sanitas SBF70) into a CSV format that openScale can import. The script handles:
1 . Reading the .xlsx file exported from the HealthCoach site
2 . Merging the Date and Time columns into openScale’s date field
3. Renaming and mapping the other columns like weight, fat, water, muscle, bone, BMI, and BMR
4. Exporting everything in the correct format to a new .csv file
 
HERE IS THE SCRIPT :

import pandas as pd
input_file = "HealthCoach.xlsx"
df = pd.read_excel(input_file)
print("Columns in Excel file:", list(df.columns))
df.columns = df.columns.str.strip()
df_openscale = pd.DataFrame()
df_openscale['date'] = pd.to_datetime(
    df['Date'].astype(str) + ' ' + df['Time'].astype(str),
    dayfirst=True, errors='coerce'
).dt.strftime('%Y-%m-%d %H:%M:%S')
df_openscale['weight'] = df.get('Weight (kg)', None)
df_openscale['fat'] = df.get('Body Fat (%)', None)
df_openscale['fat_top'] = df.get('Body Fat (Top %)', None)
df_openscale['fat_bottom'] = df.get('Body Fat (Bottom %)', None)
df_openscale['water'] = df.get('Water (%)', None)
df_openscale['muscle'] = df.get('Muscle (%)', None)
df_openscale['muscle_top'] = df.get('Muscle (Top %)', None)
df_openscale['muscle_bottom'] = df.get('Muscle (Bottom %)', None)
df_openscale['bone'] = df.get('Bone (kg)', None)
df_openscale['bmi'] = df.get('BMI', None)
df_openscale['bmr'] = df.get('BMR (kcal)', None)
df_openscale['amr'] = df.get('AMR (kcal)', None)
df_openscale['activity_level'] = df.get('Activity Level', None)
output_file = "openScale_ready_data.csv"
df_openscale.to_csv(output_file, index=False)
print(f"\n Conversion complete! File saved as: {output_file}")

Would love your feedback — if this seems helpful, I can open a pull request and contribute it directly to the repo. Let me know what you think!

oliexdev, 2025-05-23:
Thanks I added it into the wiki --> https://github.com/oliexdev/openScale/wiki/Useful-import-scripts

closed 2025-05-23 by oliexdev