oliexdev/openScale #730: Error when doing measurement Sanitas SBF 70

jobtijhuis, 2021-05-03:
**Describe the bug**
On a fresh install of the openScale Pro app use the Sanitas SBF 70 to measure and stand long enough on the scale for it to do the bodyfat measurement. This will result in an error. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start app and add profile
2. Connect scale
3. In measurement screen press Bluetooth symbol to connect
4. Step on scale to do measurement and stand on it long enough to also do body-fat measurement and such.
5. Get error

Reproduced with openScale 2.3.5-pro

**Expected behavior**
Not getting error.

**Error information**
```Build version: 2.3.5-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2021-05-03 14:37:52 
Device: HMD Global Nokia 7.2 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{73e0a41} (991ffce8-36f8-473f-80c1-4e17cb4f872d)} not attached to a context.
	at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:205)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1$1.run(BluetoothSettingsFragment.java:132)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7397)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
```

jobtijhuis, 2021-05-03:
[openScale_2021-05-03_14-43.txt](https://github.com/oliexdev/openScale/files/6414888/openScale_2021-05-03_14-43.txt)
Here is also the debug log but is was only enabled after I got the error

oliexdev, 2021-05-16:
please try out the dev version of openScale.

closed 2021-10-10 by oliexdev