oliexdev/openScale #731: Import from Apple Health

MoralCode, 2021-05-09:
**Describe the solution you'd like**
I would like to be able to import data from an apple health data export file (https://appletoolbox.com/how-to-export-apple-health-data-from-your-iphone-and-apple-watch/) into openScale.

**Describe alternatives you've considered**
None.

oliexdev, 2021-05-16:
you have to write a script to convert the xml file into a openScale csv file, see the [wikipage](https://github.com/oliexdev/openScale/wiki/Useful-import-scripts) for some other references.

MoralCode, 2022-05-28:
found [this](https://github.com/markwk/qs_ledger/tree/master/apple_health) and it seems like an excellent start (it parses the xml to a bunch of CSV files)

MoralCode, 2022-05-28:
while its sorta not a complete solution, here is a script to convert from the CSV output by the Quantified Self scripts linked above to openscale CSV format

Notes:
- only supports weight values and timestamp.
- it also adds a comment to record that it was imported from an apple health export as well as the app/version that imported it into apple health in the first place

oliexdev, 2025-05-22:
@MoralCode Do you still have the script that I can integrate it into the wikipage?

MoralCode, 2025-05-22:
Turns out i published it as a [Gist](https://gist.github.com/MoralCode/565987c78d3bfef20e453b2700a24795) and then forgot about it.

Based on the first commit message, it was based on the MiFit 'BODY' file --> openScale CSV file converter from the wiki

oliexdev, 2025-05-22:
Thanks I added it into the wiki --> https://github.com/oliexdev/openScale/wiki/Useful-import-scripts

closed 2025-05-22 by oliexdev