oliexdev/openScale #733: [FR]: Support for Children would be fine

just-Nob, 2021-05-16:
openScale is really fine for adults, but I'm using it for tracking my children's body data as well. That purpose in mind, some possibilities are missing.

What I miss most is the possibility of capturing a child's (constantly growing) height together with its weight.
There's usually also no targeted weight or height (excepted the expected height calculated out of the parents' height and the child's gender - from which again an expected average weight could be derived).
I guess that the formulas for calculating the optional weight, BMI etc. differs from those for adults as well.

Due to the many different requirements for children and adults, I think that it would be an appropriate approach to have a different set of parameters depending on the persons age, i.e. wether it's an adult or a child.

oliexdev, 2021-05-23:
I am not a medical experts.. how is the calculation differs for adults and children?

The evaluation for the weight, fat and so on depends on the age. If it can't evaluate a measurement for a specific age then openScale will not do an evaluation to avoid false information.

just-Nob, 2021-05-23:
I'm not an expert neither, but I guess the calculation differs. I will try to gather some information about that.

Anyway, one main difference between children and adults is that as an adult, the body's height usually doesn't change, but a child's height hopefully will rise.  ;-)
As a consequence, on the one hand there's no static height, but a constantly growing one. On the other hand, there's no target weight. That is both parameters (height & weight) constantly changes up to an unknown value which they approach asymptotically.
If that could be mapped in a first approach, that would be great.

oliexdev, 2021-10-10:
Yes that's a fact that kids are growing :wink: 
Unfortunately, changing the height to a non-static value will, at first glance, change a lot in the implementation but it is possible.

sserkez, 2021-12-04:
Just saving the data and displaying it would be a great first step in the implementation
Of course some adult calculations applied to kids would not make sense and this mat be ignored