oliexdev/openScale #736: Which Arduino devices are supported and how to connect them?

adams-family, 2021-05-30:
Thanks for the effort to create an open-source smart scale library.

May I suggest to document Arduino support:
- Which Arduino devices are supported?
- What is the wiring diagram?

oliexdev, 2021-06-09:
it is all explained at https://github.com/oliexdev/openScale/wiki/Custom-Bluetooth-Scale

closed 2021-06-09 by oliexdev

adams-family, 2021-06-09:
@oliexdev Thanks, I was not aware of that page, it's a very nice reading! I totally understand your motivation.

I can see that in the article above you were reverse engineering the digital display to read weight information. Have you though about reading resistance from the pressure sensors directly? I thought that it could be a generic solution that would pretty much work with all digital scales on the market. On the other hand, you would probably lose the possibility to use the built in digital display of the scale.

oliexdev, 2021-06-10:
sure you can read out the pressure sensors directly but I doubt that it is a generic solution because every scale vendor use different pressure sensors. But this solution would be nearly the same to build a scale from scratch.
You could also hack the microprocessor on the scale to get the information.

adams-family, 2021-06-10:
Yes, I think that building a scale from scratch is only hard from the mechanical point of view (nice housing, strong top lid, enclosing for electronics, etc). Thanks anyway and good luck!!