oliexdev/openScale #739: 1 by one scale, body fat calculation wrong -- female

lbdroid, 2021-06-11:
**Describe the bug**
The calculation for body fat for females is producing incorrect values. This also throws off calculations for other values that depend on body fat, such as total body water.

Value for body fat is always 45% per https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/OneByoneLib.java#L262

Suspect that bodyFatConst vals are set wrong at these two points;
https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/OneByoneLib.java#L220
https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/OneByoneLib.java#L226

Judging by the fact that for males, its set to 0.8, these should likely be set to 0.725 and 0.925 respectively.

Note that when gender for a woman is set to male in openScale, the calculated values are much better "looking", i.e., consistent with somebody who is actually ALIVE.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): ???
v 2.3.5

**Expected behavior**
Reasonable or correct values should be recorded for body fat and related fields.

lbdroid, 2021-06-12:
Here are some actual measurements using 1byOne software, and openScale. For the time being, I'm standing on it myself (male). I'll get some actual female readings once the formulas are able to generate the same data for me.

1byOne application with settings as FEMALE, DOB 01/01/1980, 5'6".
- Weight: 182.4
- BMI: 29.7
- Water: 42.2%
- Muscle: 101.4lb
- Fat: 35.0%

OpenScale with settings as FEMALE, DOB 01/01/1980, 5'6" (66"), MODERATE.
- Weight: 182.32
- BMI: 29.43
- Water: 39.27%
- Muscle: 42.54%
- Fat: 45.00%
And from the LOGS: Weight 82.70, impedanceCoeff: 14881553. From the logs also, AGE: 41, height: 167.64, activity level: 2.

lbdroid, 2021-06-13:
The problem lies here;

https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/OneByoneLib.java#L237

The block from line 237 to 253. If any of the conditions there are met, then it discards all previous calculations for bodyVar and resets the values. Based on implementation in MiScaleLib.java (which performs essentially the same calculation), it should actually be multiplying bodyVar on lines 239, 243, 247, and 251 by the applicable factors for slight adjustments rather than completely obliterating the value.

For example, line 247 should read as ```bodyVar *= 0.96f;```

oliexdev, 2021-08-21:
see PR #741

closed 2021-08-21 by oliexdev