oliexdev/openScale #740: App crash while adding Beurer BF710 scale

dewu1994, 2021-06-11:
App crashes while going to the next step after selecting Beurer BF 710. Crash logs:

`Build version: 2.3.2-dev_5d697fe9_2020-11-05 Build date: 1981-01-01 01:01:02 Current date: 2021-06-11 17:52:17 Device: Xiaomi Redmi Note 9 Pro OS version: Android 10 (SDK 29)  Stack trace:  java.lang.IllegalArgumentException: ID does not reference a View inside this Activity	at android.app.Activity.requireViewById(Activity.java:3297)	at androidx.core.app.ActivityCompat.requireViewById(ActivityCompat.java:363)	at androidx.navigation.Navigation.findNavController(Navigation.java:58)	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$BluetoothDeviceView.onClick(BluetoothSettingsFragment.java:365)	at android.view.View.performClick(View.java:7184)	at android.view.View.performClickInternal(View.java:7161)	at android.view.View.access$3500(View.java:818)	at android.view.View$PerformClick.run(View.java:27683)	at android.os.Handler.handleCallback(Handler.java:883)	at android.os.Handler.dispatchMessage(Handler.java:100)	at android.os.Looper.loop(Looper.java:224)	at android.app.ActivityThread.main(ActivityThread.java:7562)	at java.lang.reflect.Method.invoke(Native Method)	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)`

kingoffun, 2021-06-13:
Also app crash while adding this scale

Build version: 2.3.5 
Build date: 1981-01-01 01:01:02 
Current date: 2021-06-13 08:25:54 
Device: Xiaomi Mi 9 SE 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{585baa} (ba3295c9-c8a0-403f-9d3a-99fc1ae2d4a9)} not attached to a context.
	at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:205)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1$1.run(BluetoothSettingsFragment.java:132)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:236)
	at android.app.ActivityThread.main(ActivityThread.java:8061)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:656)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:967)

oliexdev, 2021-10-10:
try out the latest dev version of openScale

closed 2021-10-10 by oliexdev