oliexdev/openScale #742: Inconsistent definition of "muscle mass"

ASerbinski, 2021-06-16:
**Describe the bug**
Libraries for some scales do not calculate muscle mass correctly, resulting in provided values being too high.

The following 2 libraries, for example, use (muscle mass) = (total mass) - (fat mass) - (skeletal mass);
https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/MiScaleLib.java#L48
https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/OneByoneLib.java#L37

This is not a correct calculation, since (total weight) = (fat mass) + (lean body mass), and..
(lean body mass) = (organ mass) + (skin mass) + (bone mass) + (body water) + (muscle mass).

Consequently, what these libraries are calculating is actually "lean body mass less bone mass" and exceeds muscle mass by (organ mass) + (skin mass) + (body water). The result is that muscle mass is reported much too high, and is subject to wild swings just by having a couple of glasses of water before weighing yourself.

We can do better.

Skeletal muscle mass can actually be calculated for BIA devices through use of the Janssen formula;
```SM mass (kg) = [(Ht^2/R x 0.401) + (gender x 3.825) + (age x -0.071)] + 5.102```
Ht: height in cm,
gender: male/0, female/1,
R = BIA resistance in ohms.

*reference: https://journals.physiology.org/doi/pdf/10.1152/jappl.2000.89.2.465

There are other formulas as well, but this one seems to be the most common/popular, and is easy to implement.
Also note that this formula does not use the user's total weight as a parameter. Only height, age, gender, and impedance.

I ran this calculation on myself using the impedanceCoeff logged by openScale and converted to ohms using the OneByone formula from here; https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/OneByoneLib.java#L183 and came up with a skeletal muscle mass of 25.983 kg, which works out to 32.1%. While lower than the range offered by openScale (41-52%), it is consistent with the majority of online sources, and more importantly, academic sources (see above reference for formula, page 3), that give a range of about 33-39% (I've got a little extra padding, hence being a touch below the range).

An additional reference from the same researchers (Janssen), calculating SM mass using MRI supporting the values: https://journals.physiology.org/doi/full/10.1152/jappl.2000.89.1.81
(see fig. 2(B)).

I'll be making a pull request with this for OneByone Scale after verifying the changes this evening.

krisjans, 2021-07-24:
Hi @ASerbinski !

Found this issue while working on [Beurer BF600 support](https://github.com/oliexdev/openScale/issues/219#issuecomment-885833554)

Since you have researched body composition before, maybe you would be interested to help answer two questions below:
1. How to calculate "**bone mass**" from other measurements provided by scale (BMI, heigh, gender, age, total body weight, body fat percentage, bmr, muscle percentage, soft lean mass, body water mass and impedance);
2. Is "**Soft Lean Mass**" ( as described in [GATT_Specification_Supplement_v4](https://www.bluetooth.com/specifications/specs/gatt-specification-supplement-4/) chapter 3.27.2.5 "Soft Lean Mass field")
the same value as "**Lean body mass**" in OpenScale application?

Here is example measurements provided by my BF600 scale:
```
BMI: 21.2
heightInMeters: 1.7
weight: 61.37
bodyFatPercentage: 18.2
bmrInJoules: 6163 bmrInKcal: 1472
musclePercentage: 41.9
softLeanMass: 47.6
bodyWaterMass: 34.5
impedance: 293.7
```

ASerbinski, 2021-07-26:
This function supposedly calculates bone mass from height, impedance, and a few user set values (height, activity level, etc.): https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/OneByoneLib.java#L53

Soft lean mass is probably total_weight - fat - bone_mass.

krisjans, 2021-07-27:
@ASerbinski Thanks for your advice!

Could you provide your opinion about another question?
Does "Soft Lean Mass" (in my measurements it is `softLeanMass: 47.6`) corresponds to openScale database field "Lean body mass"?

ASerbinski, 2021-07-27:
Lean body mass includes bone mass. Soft lean mass does not. It would be best if you create your own issue for your questions.

krisjans, 2021-07-27:
> It would be best if you create your own issue for your questions.

@ASerbinski My bad! Please accept my sincere apologies!
It did not occur to me that i'm obscuring your original topic with my questions about unrelated problem! 

Would you like me to delete my questions (and this comment) from your issue (currently they only add unnecessary  noise around your original topic)?
As you suggested I opened new issue (#759) for my questions and would be extremely happy if you could answer them there.

Sorry for annoying you, it was not intentional!
Have a lucky day!
Krisjans