oliexdev/openScale #746: Wrong muscle percent normal range

Stiralbios, 2021-06-16:
The muscle percent normal range seems to be too low.
I am not muscular at all and it's in the red (see screenshot).
I found this article online https://www.menshealth.com/health/a27242669/what-your-body-composition-metrics-say-about-your-health/ where it describe what the normal percent should be. And it's higher than the one in the app.
I use the mi scale v2

Steps to reproduce the behavior:
1. Go on the mi scale v2
2. Weight yourself 
3. Activate the bluetooth it should add the new record
4. Go to the table
5. clic on the record
6. Look at the Muscle section
7. clic on it and see that's the value is in red

Reproduced with the latest fdroid version : v2.3.4 (54)

[image]

It would be nice to have the accurate normal range for the muscle percent if it's wrong, if not then maybe fix the value coming from the scale

lbdroid, 2021-06-16:
The value coming from your scale is not muscle mass. It is lean body mass less bone mass and includes other things like organs and skin mass. The scale is actually too high and should be about 31-45%.

Stiralbios, 2021-06-17:
Ok, thanks.
My description of the issue is quite wrong then. But there is still issues: 
- this value should register under lean body mass in openScale
- And, from what you said, the normal range for muscle percent seems to be to hight

lbdroid, 2021-06-17:
Well, yes and no. Its not quite lean body mass because it doesn't include bone mass, which is a component of LBM.
You can see the calculation here; https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/MiScaleLib.java#L48

There is a pull request open to fix the muscle percent scale to match academic/medical sources;
https://github.com/oliexdev/openScale/pull/748

And there is a pull request to fix the muscle calculation, provide the lean body mass to the LBM field, and correctly determine impedance (BIA), BUT NOT FOR YOUR SCALE, here;
https://github.com/oliexdev/openScale/pull/747

Now, the error for your scale is basically identical to the error for oneByone scale. If I've understood correctly, its the same manufacturer who has made the same error in their differently branded proprietary software!

Stiralbios, 2021-06-17:
Ok thanks, it's pretty cool someone already did most of the work.
I will try to understand the formula ASerbinski did and, if I am confident enough to not make mistake in the formula, do a fix for the mi scale v2.