oliexdev/openScale #750: Weekly and monthly average values not visible

iSnackyCracky, 2021-06-23:
**Describe the bug**
The weekly and monthly average graphs on the statistics tab show the average and last measurements on top of each other as a radar chart without values.
But you can't actually see the average values. If you tap one of these, it only shows the last measurement (as highlighted by the color).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Statistics'
2. Scroll down to 'weekly or monthly average'
3. Tap on the diagram

Reproduced with: `2.3.5` and also the "travis dev build" available on 2021-06-26 which shows as `v2.3.2-dev_5d697fe9_2020-11-05` in settings -> about

**Expected behavior**
Show the average values or better both the average and the last measured values.

Additionally (not really important for me) the value number 4 (bone mass) has its area in the "negative" direction (it indents towards the opposite value number 1) even though the value IS clearly positive (well, I don't think it's possible to have a negative amount of bone mass in your body 😄)

oliexdev, 2021-08-22:
true the whole statistic page needs a design review :smirk:

oliexdev, 2023-02-12:
I redesigned the statistic page see version 2.5.0

closed 2023-02-12 by oliexdev