oliexdev/openScale #751: YUNMAI Scales Bluetooth Issue

WOKNz, 2021-06-27:
The connection establishes only for a few second (Bluetooth icon appears on the scales and vanishes after just few second).  After multiple tries including force stop for the app it could stay longet and data could be transferred.

Debug Log:
`2021-06-27 09:39:00.829 Debug [2] AboutPreferences: Debug log enabled, openScale v2.3.5-pro (54), SDK 30, Xiaomi MI 8
2021-06-27 09:39:00.837 Debug [2] AboutPreferences: Selected user id(1) name(SomeName) birthday(Tue Jun 26 11:16:16 GMT+03:00 2001) age(20) body height(173.50) scale unit(kg) gender(male) initial weight(62.00) goal weight(75.00) goal date(Sun Dec 26 00:00:00 GMT+02:00 2021) measure unt(cm) activity level(1) assisted weighing(false)
2021-06-27 09:39:17.365 Info [2] BluetoothCentral: scan started
2021-06-27 09:39:37.377 Info [2] BluetoothCentral: scan stopped
2021-06-27 09:39:41.472 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-06-27 09:39:44.252 Info [2] BluetoothCentral: scan started
2021-06-27 09:39:48.105 Debug [2] BluetoothYunmaiSE_Mini: Extract the fat value from received bytes
2021-06-27 09:39:48.106 Debug [2] BluetoothYunmaiSE_Mini: received bytes [0D 1F 14 02 00 60 D8 1D 33 00 00 00 00 18 98 02 97 05 1E 91]
2021-06-27 09:39:48.107 Debug [2] BluetoothYunmaiSE_Mini: received decrypted bytes [weight: 62.96, fat: 13.10, resistance: 663]
2021-06-27 09:39:48.108 Debug [2] BluetoothYunmaiSE_Mini: user [id(1) name(SomeName) birthday(Tue Jun 26 11:16:16 GMT+03:00 2001) age(20) body height(163.50) scale unit(kg) gender(male) initial weight(62.00) goal weight(75.00) goal date(Sun Dec 26 00:00:00 GMT+02:00 2021) measure unt(cm) activity level(1) assisted weighing(false)]
2021-06-27 09:39:48.109 Debug [2] BluetoothYunmaiSE_Mini: scale measurement [ID: 0, USER_ID: -1, DATE_TIME: Sun Jun 27 09:39:47 GMT+03:00 2021, WEIGHT: 62.96, FAT: 13.10, WATER: 63.09, MUSCLE: 58.23, LBM: 54.71, WAIST: 0.00, HIP: 0.00, BONE: 2.95, CHEST: 0.00, THIGH: 0.00, ARM: 0.00, NECK: 0.00, CALIPER1: 0.00, CALIPER2: 0.00, CALIPER3: 0.00, COMMENT: ]
2021-06-27 09:39:48.115 Debug [2] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Sun Jun 27 09:39:47 GMT+03:00 2021, WEIGHT: 62.96, FAT: 13.10, WATER: 63.09, MUSCLE: 58.23, LBM: 54.71, WAIST: 0.00, HIP: 0.00, BONE: 2.95, CHEST: 0.00, THIGH: 0.00, ARM: 0.00, NECK: 0.00, CALIPER1: 0.00, CALIPER2: 0.00, CALIPER3: 0.00, COMMENT: 
2021-06-27 09:39:50.071 Info [2] BluetoothCentral: scan stopped
2021-06-27 09:40:10.469 Debug [2] BluetoothYunmaiSE_Mini: Extract the fat value from received bytes
2021-06-27 09:40:10.471 Debug [2] BluetoothYunmaiSE_Mini: received bytes [0D 1F 14 02 00 60 D8 1D 49 00 00 DA 75 18 98 02 8F 05 1E 5C]
2021-06-27 09:40:10.472 Debug [2] BluetoothYunmaiSE_Mini: received decrypted bytes [weight: 62.96, fat: 13.10, resistance: 655]
2021-06-27 09:40:10.474 Debug [2] BluetoothYunmaiSE_Mini: user [id(1) name(SomeName) birthday(Tue Jun 26 11:16:16 GMT+03:00 2001) age(20) body height(163.50) scale unit(kg) gender(male) initial weight(62.00) goal weight(75.00) goal date(Sun Dec 26 00:00:00 GMT+02:00 2021) measure unt(cm) activity level(1) assisted weighing(false)]
2021-06-27 09:40:10.475 Debug [2] BluetoothYunmaiSE_Mini: scale measurement [ID: 0, USER_ID: -1, DATE_TIME: Sun Jun 27 09:40:09 GMT+03:00 2021, WEIGHT: 62.96, FAT: 13.10, WATER: 63.09, MUSCLE: 58.23, LBM: 54.71, WAIST: 0.00, HIP: 0.00, BONE: 2.95, CHEST: 0.00, THIGH: 0.00, ARM: 0.00, NECK: 0.00, CALIPER1: 0.00, CALIPER2: 0.00, CALIPER3: 0.00, COMMENT: ]
2021-06-27 09:40:10.481 Debug [2] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Sun Jun 27 09:40:09 GMT+03:00 2021, WEIGHT: 62.96, FAT: 13.10, WATER: 63.09, MUSCLE: 58.23, LBM: 54.71, WAIST: 0.00, HIP: 0.00, BONE: 2.95, CHEST: 0.00, THIGH: 0.00, ARM: 0.00, NECK: 0.00, CALIPER1: 0.00, CALIPER2: 0.00, CALIPER3: 0.00, COMMENT: 
2021-06-27 09:40:10.548 Debug [2] BluetoothYunmaiSE_Mini: Extract the fat value from received bytes
2021-06-27 09:40:10.549 Debug [2] BluetoothYunmaiSE_Mini: received bytes [0D 1F 14 02 00 60 D8 1D 49 00 00 DA 75 18 98 02 8F 05 1E 5C]
2021-06-27 09:40:10.549 Debug [2] BluetoothYunmaiSE_Mini: received decrypted bytes [weight: 62.96, fat: 13.10, resistance: 655]
2021-06-27 09:40:10.550 Debug [2] BluetoothYunmaiSE_Mini: user [id(1) name(SomeName) birthday(Tue Jun 26 11:16:16 GMT+03:00 2001) age(20) body height(163.50) scale unit(kg) gender(male) initial weight(62.00) goal weight(75.00) goal date(Sun Dec 26 00:00:00 GMT+02:00 2021) measure unt(cm) activity level(1) assisted weighing(false)]
2021-06-27 09:40:10.550 Debug [2] BluetoothYunmaiSE_Mini: scale measurement [ID: 0, USER_ID: -1, DATE_TIME: Sun Jun 27 09:40:09 GMT+03:00 2021, WEIGHT: 62.96, FAT: 13.10, WATER: 63.09, MUSCLE: 58.23, LBM: 54.71, WAIST: 0.00, HIP: 0.00, BONE: 2.95, CHEST: 0.00, THIGH: 0.00, ARM: 0.00, NECK: 0.00, CALIPER1: 0.00, CALIPER2: 0.00, CALIPER3: 0.00, COMMENT: ]
2021-06-27 09:40:10.558 Debug [2] OpenScale: to be added measurement is thrown away because measurement with the same date and time already exist
2021-06-27 09:41:33.028 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect
2021-06-27 09:41:33.035 Debug [2] BluetoothCommunication: Bluetooth disconnect
2021-06-27 09:41:33.045 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-06-27 09:41:33.079 Debug [2] MainActivity: Bluetooth connection successful disconnected
2021-06-27 09:41:33.083 Info [2] BluetoothPeripheral: force disconnect 'YUNMAI-ISM2-W' (6C:C3:74:D4:37:BB)
2021-06-27 09:41:33.092 Info [15530] BluetoothPeripheral: disconnected 'YUNMAI-ISM2-W' on request
2021-06-27 09:41:33.120 Debug [2] BluetoothCommunication: disconnected 'YUNMAI-ISM2-W' with status 0
`

WOKNz, 2021-06-28:
After the phone goes sleep in this "connected" state, I unlock the phone and turn off the connection via the app and turn on again, it finally connects for longer than 2sec.

oliexdev, 2021-08-21:
referring to your logs two measurements were added and one were thrown away because it was already added.

closed 2021-08-21 by oliexdev