oliexdev/openScale #752: [Question] Any workaround to pair the Fitindex ES-26M?

JamoDevNich, 2021-07-03:
I have a FitIndex ES-26M (identical to the one shown on the [supported scales wiki](https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale) page)  

Sadly the scale does not broadcast a device name, so cannot be paired with openScale, even though it appears to be supported according to the wiki. For clarification, this scale is new out of the box and has not been paired to anything. There are no buttons on it either  

The scale may have broadcast a name in the past, however it is possible a hardware or software revision has changed this. It may also be possible that during development the scale was already paired to the app. Sadly none of the issues mentioning this scale (at time of writing) contain a solution on how to pair it to openScale.  

**Is there a way to manually add this scale to openScale, e.g. export/backup the database via ADB and add an entry with the MAC address by hand?**  

The rest of this post is just additional discoveries, debug log, screenshots, and some ideas for potential fixes

&nbsp;  

## Code

I'm not an Android/Java dev so this is just an assumption; a section of the bluetooth discovery code appears to exclude devices with no broadcast name  

https://github.com/oliexdev/openScale/blob/77dd3227c3c2ed75805d532cd3c3bab3703ff90c/android_app/app/src/main/java/com/health/openscale/gui/preferences/BluetoothSettingsFragment.java#L198-L203  

&nbsp;  

## Logs

Below is a scan in openScale with debug logging enabled, I redacted some personal information, and several sensor devices

```txt
2021-07-03 14:54:06.767 Debug [1] AboutPreferences: Debug log enabled, openScale v2.3.5 (54), SDK 21, samsung SM-G900F
2021-07-03 14:54:06.784 Debug [1] AboutPreferences: Selected user id(1) name(<Redacted>) birthday(<Redacted>) age(<Redacted>) body height(<Redacted>) scale unit(kg) gender(male) initial weight(<Redacted>) goal weight(<Redacted>) goal date(Wed Dec 28 00:00:00 GMT+00:00 2022) measure unt(cm) activity level(2) assisted weighing(false)
2021-07-03 14:54:15.641 Info [1] BluetoothCentral: scan started
2021-07-03 14:54:16.916 Debug [1] BluetoothSettingsFragme: Found unsupported device LYWSD03MMC [A4:C1:38:<Redacted>]
2021-07-03 14:54:16.927 Debug [1] BluetoothSettingsFragme: Found unsupported device LYWSD03MMC [A4:C1:38:<Redacted>]
2021-07-03 14:54:17.903 Debug [1] BluetoothSettingsFragme: Found unsupported device LYWSD03MMC [A4:C1:38:<Redacted>]
2021-07-03 14:54:35.677 Info [1] BluetoothCentral: scan stopped
2021-07-03 14:54:40.948 Info [1] BluetoothCentral: no scan to stop because no scan is running
2021-07-03 14:54:49.467 Debug [1] AboutPreferences: Debug log disabled
```

Just in-case, I also tried the latest dev version from the release page

```txt
2021-07-03 16:28:59.503 Debug [1] AboutPreferences: Debug log enabled, openScale (dev) v2.3.2-dev_5d697fe9_2020-11-05 (1604584993), SDK 21, samsung SM-G900F
2021-07-03 16:28:59.505 Debug [1] AboutPreferences: Selected user id(0) name() birthday(Sat Jul 03 16:28:59 BST 2021) age(0) body height(-1.00) scale unit(kg) gender(male) initial weight(-1.00) goal weight(-1.00) goal date(Sat Jul 03 16:28:59 BST 2021) measure unt(cm) activity level(0) assisted weighing(false)
2021-07-03 16:29:45.110 Info [1] BluetoothCentral: scan started
2021-07-03 16:29:46.066 Debug [1] BluetoothSettingsFragme: Found unsupported device LYWSD03MMC [A4:C1:38:<redacted>]
2021-07-03 16:29:46.551 Debug [1] BluetoothSettingsFragme: Found unsupported device LYWSD03MMC [A4:C1:38:<redacted>]
2021-07-03 16:29:51.749 Debug [1] BluetoothSettingsFragme: Found unsupported device LYWSD03MMC [A4:C1:38:<redacted>]
2021-07-03 16:30:05.134 Info [1] BluetoothCentral: scan stopped
2021-07-03 16:31:01.387 Info [1] BluetoothCentral: no scan to stop because no scan is running
2021-07-03 16:31:09.289 Debug [1] AboutPreferences: Debug log disabled

```

&nbsp;  

## Screenshots

The scale shows up in the stock Android bluetooth scanner (with no name), and also in a third party bluetooth terminal application (with no name). The MAC address is `ED:67:38:5E:27:5F`  

<details>
<summary>Expand to show device screenshots</summary>

[image]

[image]

</details>

&nbsp;  

## Ideas on how to fix  

1. If the manufacturer is using the same first three octets of the MAC address for this scale model, it may be possible to...  
    - allow the discovery code to present the first three MAC address octets as the device name - this way, the existing 'supported scales' checker can be kept without modification, an entry for the first three octets would be added as the device name for this particular scale
    - instead of the above, using an internal lookup table to assign a name based on the first three octets (manufacturer) 
2. In-app manually enter the MAC address and a scale name (to be used to detect the correct implementation)  
3. Some other workaround to manually provide the scale information to the app, even if it's importing a text file with a name and MAC address  

None of these are ideal to be honest, it would have been awesome if the scales just sent a device name to begin with. 

Do you have any ideas how this could be fixed, or if it is feasible to?

JamoDevNich, 2021-07-03:
I managed to add the scale (backup over ADB; added two additional entries to the preferences.xml after a quick look at the bluetooth code here).  

<details>
<summary>Expand for extraction details</summary>

1. Pulled app backup from the device: `adb backup com.health.openscale`
2. Extracted backup file to a tar file: `tail -c +25 backup.ab | openssl zlib -d > backup.ab.tar`
3. Open `backup.ab.tar` in an archive browser of your choice 
4. Navigate to the `com.health.openscale_preferences.xml` file and open it in 'edit mode'. Add the following lines inside the `<map>` tags:
    ```xml
    <string name="btDeviceName">QN-Scale</string>
    <string name="btHwAddress">MAC:ADDRESS:GOES:HERE</string>
    ```
5. Save the changes
6. Repack the archive: `head -c +24 backup.ab > backup.repacked.ab && openssl zlib -in backup.ab.tar >> backup.repacked.ab`
7. Push the app backup to the device: `adb restore backup.repacked.ab`

</details>

In the logs it successfully identifies the scale with the MAC address, but fails to connect. I've attached them below

```txt
2021-07-03 19:45:25.144 Debug [1] AboutPreferences: Debug log enabled, openScale v2.3.5 (54), SDK 21, samsung SM-G900F
2021-07-03 19:45:25.146 Debug [1] AboutPreferences: Selected user id(1) name(Nich) birthday(<Redacted>) age(<Redacted>) body height(<Redacted>) scale unit(kg) gender(male) initial weight(<Redacted>) goal weight(<Redacted>) goal date(Wed Dec 27 00:00:00 GMT+00:00 2023) measure unt(cm) activity level(2) assisted weighing(false)
2021-07-03 19:45:43.477 Debug [1] OpenScale: Disconnecting from bluetooth device
2021-07-03 19:45:43.486 Debug [1] BluetoothCommunication: Bluetooth disconnect
2021-07-03 19:45:43.487 Info [1] BluetoothCentral: no scan to stop because no scan is running
2021-07-03 19:45:43.515 Debug [1] MainActivity: Bluetooth connection successful disconnected
2021-07-03 19:45:45.343 Debug [1] OpenScale: Trying to connect to bluetooth device [ED:67:38:5E:27:5F] (QN-Scale)
2021-07-03 19:45:45.347 Debug [1] BluetoothCommunication: Do LE scan before connecting to device
2021-07-03 19:45:45.364 Info [1] BluetoothCentral: scan started
2021-07-03 19:45:45.375 Debug [1] BluetoothCommunication: Stop machine state
2021-07-03 19:45:45.432 Debug [1] BluetoothCommunication: Found peripheral 'null'
2021-07-03 19:45:45.450 Info [1] BluetoothCentral: scan stopped
2021-07-03 19:45:46.455 Debug [1] BluetoothCommunication: Try to connect to BLE device ED:67:38:5E:27:5F
2021-07-03 19:45:46.575 Info [1] BluetoothPeripheral: connect to 'null' (ED:67:38:5E:27:5F) using TRANSPORT_LE
2021-07-03 19:45:51.638 Info [41825] BluetoothPeripheral: connection failed with status 'GATT_ERROR' (TIMEOUT)
2021-07-03 19:45:51.653 Info [41825] BluetoothCentral: connection to 'null' (ED:67:38:5E:27:5F) failed
2021-07-03 19:45:51.654 Error [1] BluetoothCommunication: connection 'null' failed with status 8
2021-07-03 19:45:51.660 Debug [1] MainActivity: Bluetooth connection lost
2021-07-03 19:45:51.665 Debug [1] MainActivity: Bluetooth scale message: Could not connect to scale, please ensure it is on.
2021-07-03 19:46:00.368 Debug [1] OpenScale: Disconnecting from bluetooth device
2021-07-03 19:46:00.371 Debug [1] BluetoothCommunication: Bluetooth disconnect
2021-07-03 19:46:00.377 Info [1] BluetoothCentral: no scan to stop because no scan is running
2021-07-03 19:46:00.412 Debug [1] MainActivity: Bluetooth connection successful disconnected
2021-07-03 19:46:06.315 Debug [1] OpenScale: Trying to connect to bluetooth device [ED:67:38:5E:27:5F] (QN-Scale)
2021-07-03 19:46:06.327 Debug [1] BluetoothCommunication: Do LE scan before connecting to device
2021-07-03 19:46:06.340 Info [1] BluetoothCentral: scan started
2021-07-03 19:46:06.343 Debug [1] BluetoothCommunication: Stop machine state
2021-07-03 19:46:06.702 Debug [1] BluetoothCommunication: Found peripheral 'null'
2021-07-03 19:46:06.719 Info [1] BluetoothCentral: scan stopped
2021-07-03 19:46:07.727 Debug [1] BluetoothCommunication: Try to connect to BLE device ED:67:38:5E:27:5F
2021-07-03 19:46:07.829 Info [1] BluetoothPeripheral: connect to 'null' (ED:67:38:5E:27:5F) using TRANSPORT_LE
2021-07-03 19:46:12.854 Info [41825] BluetoothPeripheral: connection failed with status 'GATT_ERROR' (TIMEOUT)
2021-07-03 19:46:12.861 Info [41825] BluetoothCentral: connection to 'null' (ED:67:38:5E:27:5F) failed
2021-07-03 19:46:12.869 Error [1] BluetoothCommunication: connection 'null' failed with status 8
2021-07-03 19:46:12.888 Debug [1] MainActivity: Bluetooth connection lost
2021-07-03 19:46:12.899 Debug [1] MainActivity: Bluetooth scale message: Could not connect to scale, please ensure it is on.
2021-07-03 19:46:25.624 Debug [1] OpenScale: Disconnecting from bluetooth device
2021-07-03 19:46:25.632 Debug [1] BluetoothCommunication: Bluetooth disconnect
2021-07-03 19:46:25.633 Info [1] BluetoothCentral: no scan to stop because no scan is running
2021-07-03 19:46:25.654 Debug [1] MainActivity: Bluetooth connection successful disconnected
2021-07-03 19:46:34.649 Debug [1] AboutPreferences: Debug log disabled
```

I think/assume this could be a new hardware/software revision of the scale and some commands may have changed, considering the amount of recent issues for other scales using the QTScale implementation, as well as the scales no longer broadcasting a device name.  

If I keep these scales I'll likely try to reverse engineer them when I can find time.

closed 2021-07-03 by JamoDevNich

JamoDevNich, 2021-07-04:
Update: Reverse engineered the scales and got them working in the app! 

They seemed mostly compatible with the "OKOK" and "QNScale" libraries, so those were the basis for a new driver created for these scales. I've uploaded the new driver implementation to a new branch for the time being if anyone wants to have a look or build it.  

Steps mentioned in my previous post are a prerequisite to add the scales though... because they do not broadcast a name they can't be added via the in-app bluetooth scanning screen. In the preferences XML file, use `ESScale-Rev2` as the device name.  

Reverse engineering notes below  

---

The measurements are found in the manufacturer data record `0xFFFF` and broadcast several times a second from the scales.

### Empty scales
*Measurements taken between power-off/calibration cycles*  
Weight: 0.00KG (*From built in display*)
```
[-86, -69, -19, 103, 56, 94, 39, 95,  56, 100, 1, 0, -1, -1, -1, 2, 0, 0, 0, 73, 89, 3,  96, 9]
[-86, -69, -19, 103, 56, 94, 39, 95, 107, 103, 1, 0, -1, -1, -1, 2, 0, 0, 0, 73, 89, 3, -52, 9]
[-86, -69, -19, 103, 56, 94, 39, 95, -85, 103, 1, 0, -1, -1, -1, 2, 0, 0, 0, 73, 89, 3, -26, 9]
[-86, -69, -19, 103, 56, 94, 39, 95,  20, 104, 1, 0, -1, -1, -1, 2, 0, 0, 0, 73, 89, 3,  2, 10]
```

### On scales: Not on impedance pads, weight fluctuating/unstable/incomplete
*Measurements taken in the same power cycle i.e. scales not powered down or allowed to calibrate between measurements*
Weight: <8KG (*Estimated, Fluctuating*)
```
[-86, -69, -19, 103, 56, 94, 39, 95, -112, 105, 1, 0, -1, -1, -1, 2, 0,  -7, 1, 73, 89, 3, 24, 10]
[-86, -69, -19, 103, 56, 94, 39, 95, -110, 105, 1, 0, -1, -1, -1, 2, 0, -32, 1, 73, 89, 3, 28, 10]
[-86, -69, -19, 103, 56, 94, 39, 95, -109, 105, 1, 0, -1, -1, -1, 2, 0, -83, 2, 73, 89, 3, 29, 10]
[-86, -69, -19, 103, 56, 94, 39, 95, -109, 105, 1, 0, -1, -1, -1, 2, 0, 118, 2, 73, 89, 3, 29, 10]
[-86, -69, -19, 103, 56, 94, 39, 95, -109, 105, 1, 0, -1, -1, -1, 2, 0,  63, 2, 73, 89, 3, 29, 10]
                                                                         ^   ^
                                                                Weight LSB   Weight MSB
```

### On scales: Not on impedance pads, weight complete
*Measurements taken in the same power cycle i.e. scales not powered down or allowed to calibrate between measurements*
Weight: 8.25KG (*From built in display*)
```
[-86, -69, -19, 103, 56, 94, 39, 95, -94, 106, 1, 0, -1, -1, -1, 2, 0, 57, 3, 73, 89, 3, 116, 10]
[-86, -69, -19, 103, 56, 94, 39, 95, -92, 106, 1, 0, -1, -1, -1, 2, 0, 57, 3, 73, 89, 3, 121, 10]
```

### On scales: On impedance pads, weight complete, impedance not complete
*Measurements taken in the same power cycle i.e. scales not powered down or allowed to calibrate between measurements*
Weight: 80.85KG (*From built in display*)
```
[-86, -69, -19, 103, 56, 94, 39, 95, -80, 109, 1, 0, -1, -1, -1, 2, 0, -107, 31, 73, 89, 3, 66, 11]
[-86, -69, -19, 103, 56, 94, 39, 95, -80, 109, 1, 0, -1, -1, -1, 2, 0, -107, 31, 73, 89, 3, 67, 11]
```

### On scales: On impedance pads, weight complete, impedance complete
*Measurements taken in the same power cycle i.e. scales not powered down or allowed to calibrate between measurements*
Weight: 81.70KG (*From built in display*)
```
[-86, -69, -19, 103, 56, 94, 39, 95, 117, 110, 1, 0, -1, -1, -1, 35, 0, -22, 31, 73, 89, 3, 110, 11]
[-86, -69, -19, 103, 56, 94, 39, 95, 118, 110, 1, 0, -1, -1, -1, 35, 0, -22, 31, 73, 89, 3, 110, 11]
[-86, -69, -19, 103, 56, 94, 39, 95, 119, 110, 1, 0, -1, -1, -1, 35, 0, -22, 31, 73, 89, 3, 110, 11]
[-86, -69, -19, 103, 56, 94, 39, 95, 120, 110, 1, 0, -1, -1, -1, 35, 0, -22, 31, 73, 89, 3, 110, 11]
                                      ^                          ^                           ^   ^
                                      |                          |                Impedance LSB  |
                        Unknown self-incrementing value          |                    Impedance MSB
                                                      complete indicator
```

Plokmju22, 2022-04-23:
I think you're on to something about about the first 3 of the MAC address - I'm have the same issue on mine and the MAC address is **ED:67:38:F4:F2:33**.