oliexdev/openScale #755: No data transferred from Sanitas SBF70

charlydelta, 2021-07-20:
**Describe the bug**
I bought a Sanitas SBF70 a few weeks ago, however I just cannot make the scale transfer data to my device. It shows the bluetooth logo, and I also get toasts on my deivce like the currently measured weight, and the "connected" infos etc., but nothing else is transmitted yet saved. I setup the profile as suggested in the documentation. I don't have Google Play store so I cannot test the manufacturers app.

**To Reproduce**
Steps to reproduce the behavior:
1. Open openScale (auto-connect on startup enabled)
2. Step on scale
3. "connected"-toast
4. Nothing happens except the currently measured weight as toasts

**Expected behavior**
I expect data to be transmitted to my device, i.e. weight, fat%, water%, ...

**Additional context**
- I am using the current version available on fdroid.
- The name in the app and the name indicated on the scale do match (user setup as suggested in documentation)

**Debug log**
How I obtained the logs (maybe I do something wrong)

- Opened the app
- Opened settings and enabled save logs to file
- Disconnected bluetooth and connected again
- After connection established (bluetooth logo was visible on scale) I stepped up
- Weight readings both on scale and device match (scale recognizes my profile)
- Nothing else happens
- Scale turns off
- Bluetooth connection closes

```
2021-07-20 09:25:08.387 Debug [2] AboutPreferences: Debug log enabled, openScale v2.3.5 (54), SDK 29, Fairphone FP3
2021-07-20 09:25:08.391 Debug [2] AboutPreferences: Selected user id(5) name(REDACTED) birthday(REDACTED) age(REDACTED) body height(REDACTED) scale unit(REDACTED) gender(REDACTED) initial weight(REDACTED) goal weight(REDACTED) goal date(REDACTED) measure unt(REDACTED) activity level(REDACTED) assisted weighing(REDACTED)
2021-07-20 09:25:21.082 Debug [2] OpenScale: Disconnecting from bluetooth device
2021-07-20 09:25:21.084 Debug [2] BluetoothCommunication: Bluetooth disconnect
2021-07-20 09:25:21.086 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-07-20 09:25:21.096 Debug [2] MainActivity: Bluetooth connection successful disconnected
2021-07-20 09:25:23.166 Debug [2] OpenScale: Trying to connect to bluetooth device [78:A5:04:69:E6:A7] (SANITAS SBF70)
2021-07-20 09:25:23.169 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2021-07-20 09:25:23.184 Info [2] BluetoothCentral: scan started
2021-07-20 09:25:23.185 Debug [2] BluetoothCommunication: Stop machine state
2021-07-20 09:25:23.418 Debug [2] BluetoothCommunication: Found peripheral 'SANITAS SBF70'
2021-07-20 09:25:23.423 Info [2] BluetoothCentral: scan stopped
2021-07-20 09:25:24.430 Debug [2] BluetoothCommunication: Try to connect to BLE device 78:A5:04:69:E6:A7
2021-07-20 09:25:24.545 Info [2] BluetoothPeripheral: connect to 'SANITAS SBF70' (78:A5:04:69:E6:A7) using TRANSPORT_LE
2021-07-20 09:25:25.023 Info [1947] BluetoothPeripheral: connected to 'SANITAS SBF70' (BOND_NONE) in 0.5s
2021-07-20 09:25:25.029 Debug [2] BluetoothPeripheral: discovering services of 'SANITAS SBF70' with delay of 0 ms
2021-07-20 09:25:26.203 Info [1947] BluetoothPeripheral: discovered 5 services for 'SANITAS SBF70'
2021-07-20 09:25:26.208 Debug [2] BluetoothCommunication: connected to 'SANITAS SBF70'
2021-07-20 09:25:26.209 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2021-07-20 09:25:26.211 Debug [2] BluetoothCommunication: Resume machine state
2021-07-20 09:25:26.212 Debug [2] BluetoothCommunication: Step Nr 0
2021-07-20 09:25:26.213 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2021-07-20 09:25:26.214 Debug [2] BluetoothCommunication: Stop machine state
2021-07-20 09:25:26.234 Debug [2] MainActivity: Bluetooth connection successful established
2021-07-20 09:25:26.346 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2021-07-20 09:25:26.347 Debug [2] BluetoothCommunication: Resume machine state
2021-07-20 09:25:26.347 Debug [2] BluetoothCommunication: Step Nr 1
2021-07-20 09:25:26.348 Debug [2] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2021-07-20 09:25:26.349 Debug [2] BluetoothCommunication: Stop machine state
2021-07-20 09:25:26.352 Debug [2] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-07-20 09:25:26.496 Debug [2] BluetoothCommunication: SUCCESS: Writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-07-20 09:25:26.498 Debug [2] BluetoothBeurerSanitas: Got init ack from scale; scale is ready
2021-07-20 09:25:26.498 Debug [2] BluetoothCommunication: Resume machine state
2021-07-20 09:25:26.499 Debug [2] BluetoothCommunication: Step Nr 2
2021-07-20 09:25:26.500 Debug [2] BluetoothCommunication: Invoke write bytes [E9 60 F6 7A 66] on 0xffe1
2021-07-20 09:25:26.501 Debug [2] BluetoothCommunication: Step Nr 3
2021-07-20 09:25:26.501 Debug [2] BluetoothCommunication: Invoke write bytes [E7 4F 00 00 00 00 00 00 00 00] on 0xffe1
2021-07-20 09:25:26.502 Debug [2] BluetoothCommunication: Stop machine state
2021-07-20 09:25:26.505 Debug [2] BluetoothPeripheral: writing <e960f67a66> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-07-20 09:25:26.590 Debug [2] BluetoothCommunication: SUCCESS: Writing <E9 60 F6 7A 66> to <0000ffe1-0000-1000-8000-00805f9b34fb>
[... 26 lines cut ...]
2021-07-20 09:25:35.681 Debug [2] MainActivity: Bluetooth scale message: Measuring weight: 82.40
2021-07-20 09:25:37.034 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 82.40
2021-07-20 09:25:37.053 Debug [2] MainActivity: Bluetooth scale message: Measuring weight: 82.40
2021-07-20 09:25:38.180 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 82.45
2021-07-20 09:25:38.195 Debug [2] MainActivity: Bluetooth scale message: Measuring weight: 82.45
2021-07-20 09:25:39.438 Debug [2] BluetoothBeurerSanitas: Active measurement, weight: 82.40
2021-07-20 09:25:39.456 Debug [2] MainActivity: Bluetooth scale message: Measuring weight: 82.40
2021-07-20 09:25:40.933 Info [2] BluetoothBeurerSanitas: Active measurement, stable weight: 82.40
2021-07-20 09:25:42.128 Info [2] BluetoothBeurerSanitas: Active measurement, stable weight: 82.40
2021-07-20 09:25:47.786 Debug [2] BluetoothCommunication: Invoke write bytes [E7 F1 59 03 01] on 0xffe1
2021-07-20 09:25:47.796 Debug [2] BluetoothPeripheral: writing <e7f1590301> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-07-20 09:25:47.884 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 F1 59 03 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-07-20 09:25:47.888 Debug [2] BluetoothCommunication: Invoke write bytes [E7 F1 59 03 02] on 0xffe1
2021-07-20 09:25:47.899 Debug [2] BluetoothPeripheral: writing <e7f1590302> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-07-20 09:25:48.034 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 F1 59 03 02> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-07-20 09:26:47.903 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect
2021-07-20 09:26:47.904 Debug [2] BluetoothCommunication: Bluetooth disconnect
2021-07-20 09:26:47.905 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-07-20 09:26:47.924 Debug [2] MainActivity: Bluetooth connection successful disconnected
2021-07-20 09:26:47.927 Info [2] BluetoothPeripheral: force disconnect 'SANITAS SBF70' (78:A5:04:69:E6:A7)
2021-07-20 09:26:47.936 Info [1936] BluetoothPeripheral: disconnected 'SANITAS SBF70' on request
2021-07-20 09:26:47.970 Debug [2] BluetoothCommunication: disconnected 'SANITAS SBF70' with status 0
```

As long as I am not doing something horribly wrong, I don't think that this is particularly helpful for you to debug, so please let me know what else I can do/provide.

krisjans, 2021-07-24:
@charlydelta Hi!
Without **btsnoop_hci.log** from original app it will be impossible to tell how to communicate with this scale.
Did you checked apk mirror sites?

charlydelta, 2021-07-24:
> @charlydelta Hi!
> Without **btsnoop_hci.log** from original app it will be impossible to tell how to communicate with this scale.
> Did you checked apk mirror sites?

But the scale is listed as a supported scale, why is it then needed to know how to communicate with it? Wasn't it already reverse-engineered?

krisjans, 2021-07-24:
> But the scale is listed as a supported scale, why is it then needed to know how to communicate with it? Wasn't it already reverse-engineered?

This scale might have newer hardware or firmware (most likely both), but the scale name was not changed by manufacturer (for marketing purposes for example). There also might be a bug in openScale.
Either way - **btsnoop_hci.log** from original app is needed to clarify that.

charlydelta, 2021-07-24:
> This scale might have newer hardware or firmware (most likely both), but the scale name was not changed by manufacturer (for marketing purposes for example). There also might be a bug in openScale. [...]

All right, thanks. I will try to provide the needed files asap. Thanks for your help :)

charlydelta, 2021-08-01:
I now installed HealthCoach and reset the scale. Even though I can configure my user and everything, I just cannot fetch any data other than the weight from the scale with either app. But there doesn't seem to be any error, since neither app reports anything. So even if I open HealthCoach and step on the scale no data will be transferred. I don't know if I just do anything wrong the whole time or if the scale is broken somehow.

I tried to snoop the BT connection, but even after hours of trial and error and reading tutorials and stuff, my phone will just not generate a log file...

krisjans, 2021-08-03:
> I tried to snoop the BT connection, but even after hours of trial and error and reading tutorials and stuff, my phone will just not generate a log file...

@charlydelta what is your phone model and Android version?

p.s. If you have older phone (or older tablet) with older Android version then try it. Procedure to generate btsnoop_hci.log on newest Android version is a bit more tricky than on older versions.

charlydelta, 2021-08-03:
Yes I've got a Fairphone 3+ with latest LineageOS. I don't have an older device uinfortunately. But maybe I can ask a buddy, I will see tomorrow.

Alfred19, 2021-08-11:
Oi lads and lasses, i suffered from the same problem and i may have found a solution. 

Charlydelta: bad news for you have to get Santitas Health Coach app to create a user (this step was somewhere in the 'how to' a time ago(we need a proper manual for the thing)) you may want to get it Aurora store app with spoofed google ID.
Then i downgraded to 2.3.4 and everything seems to be working just fine again.
Running LineageOS 16 without .gapps on a Moto G7 US version if that helps.

If it changes for some reason ill let you know.

new-on-github, 2022-01-24:
Hello,

I have also a problem. But I'm not sure if it is the same.

I can access my Santitas SBF70 with one phone but not with the other.
Works fine:
Note 4 with LineageOS Android 9
App: 2.4.5 (fdroid)
Cannot connect:
Pixel 3 XL CalyxOS Android 11
App. 2.4.5 (fdroid)

charlie89, 2022-05-08:
I also experience this bug when using the latest version (and older ones) from F-Droid. But when i manually compile and install openScale then it just works and it shows all the measurements i've done.
So it seems there is something different in the debug version or in the way you guys compile the apk for the F-Droid store...?

Steps i have done: clone this repository, open the android_app directory with android studio, select Build > Build Bundle(s) / APK(s) > Build APK(s), copy the file android_app/app/build/outputs/apk/debug/openScale-debug.apk to my phone and install it.
Hardware: Sanitas SBF70, Poco F2 Pro with MIUI Global 13.0.2 (Android 12).

coxtor, 2022-07-25:
Bump, i have the same issue

charlydelta, 2023-11-01:
A minor update from my side: Meanwhile I bought a new phone, and the same issues still persist. With the new phone I also have Google Play so I now also tested the manufacturers app. I cannot get it to work with that either, so I assume that at least for my case, it is just a broken scale...