oliexdev/openScale #760: Support metric Units

cocker-cc, 2021-07-28:
**Is your feature request related to a problem? Please describe.**
Living in a modern and developed World, I am always frustrated when I have to deal with non-metric Units.

**Describe the solution you'd like**
I'd like to be able to choose to display metric Units:

- [x] `label_weight`: `kg`
- [ ] `label_bmi`: `kg/m²`
- [ ] `label_bmr`: `kJ/d` and `W` (choosable)
- [ ] `label_tdee`: `kJ/d` and `W` (choosable)
- [ ] `label_calories`: `kJ`

For `label_calories` please use the String `Energy` and not `Calories`: not to confuse physical Quantity and Unit of Measurement.

sarika-03, 2025-05-21:
Hi, can I work on this issue?

oliexdev, 2025-05-22:
sure go for it

sarika-03, 2025-07-13:
Hi @oliexdev 👋

Thanks again for giving me response . 
To make sure I implement it correctly, could you please guide me on:

1. Where are the unit labels (like `label_weight`, `label_bmr`, etc.) currently defined?
2. Is there any existing logic for unit conversion in the codebase?
3. Should I add a user setting to switch between metric and non-metric units, or just show metric units for now?

This will help me make accurate changes. Appreciate your time!