oliexdev/openScale #761: Crash on add Bluetooth scale on first install

tetsuo55, 2021-07-28:
**Describe the bug**
Crash when trying to add Bluetooth scale in setup screen

**To Reproduce**
Steps to reproduce the behavior:
1. Go to add Bluetooth scale on clean install
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 2.3.2-dev_5d697fe9_2020-11-05

**Expected behavior**
No crashing 

**Additional context**
I'm adding an AEG branded Exingtech Y1

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

Build version: 2.3.2-dev_5d697fe9_2020-11-05 
Build date: 1981-01-01 01:01:02 
Current date: 2021-07-28 19:48:20 
Device: Xiaomi Mi 9T Pro 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{264da53} (9d5419d9-364c-4cb8-aaed-d2f5e0ae510d)} not attached to a context.
	at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:201)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1$1.run(BluetoothSettingsFragment.java:132)
	at android.os.Handler.handleCallback(Handler.java:914)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:224)
	at android.app.ActivityThread.main(ActivityThread.java:7560)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)

tetsuo55, 2021-07-28:
When the crash above doesn't occur i get this one upon selecting the scale from the list of devices

Build version: 2.3.2-dev_5d697fe9_2020-11-05 
Build date: 1981-01-01 01:01:02 
Current date: 2021-07-28 19:50:00 
Device: Xiaomi Mi 9T Pro 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.IllegalArgumentException: ID does not reference a View inside this Activity
	at android.app.Activity.requireViewById(Activity.java:3299)
	at androidx.core.app.ActivityCompat.requireViewById(ActivityCompat.java:363)
	at androidx.navigation.Navigation.findNavController(Navigation.java:58)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$BluetoothDeviceView.onClick(BluetoothSettingsFragment.java:365)
	at android.view.View.performClick(View.java:7252)
	at android.view.View.performClickInternal(View.java:7229)
	at android.view.View.access$3500(View.java:821)
	at android.view.View$PerformClick.run(View.java:27759)
	at android.os.Handler.handleCallback(Handler.java:914)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:224)
	at android.app.ActivityThread.main(ActivityThread.java:7560)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)

oliexdev, 2021-08-22:
please try out the latest dev version, this error should be fixed.

closed 2021-08-22 by oliexdev

krisjans, 2021-08-25:
I think @bcdiaconu found another scenario how re-create this crash, see comment https://github.com/oliexdev/openScale/issues/502#issuecomment-905178179 and https://github.com/oliexdev/openScale/issues/502#issuecomment-905433896.

tetsuo55, 2021-08-25:
> I think @bcdiaconu found another scenario how re-create this crash, see comment [#502 (comment)](https://github.com/oliexdev/openScale/issues/502#issuecomment-905178179) and [#502 (comment)](https://github.com/oliexdev/openScale/issues/502#issuecomment-905433896).

Thanks @krisjans, your latest build solves all my issues and for the first time i got impedance sensor data

krisjans, 2021-08-28:
Finally I managed to repeat the crash.
Approximate recreation scenario:

1. uninstall openScale
2. install openScale
3. open openScale
4. go trough tutorial
5. create user while in tutorial
6. connect to scale in tutorial, but be quick about it
7. quickly close tutorial with done button
8. rotate phone? not sure about this step
9. sometimes app crashes
 
```
Build version: 2.3.5 
Build date: 1981-01-01 01:01:02 
Current date: 2021-08-28 21:08:21 
Device: Samsung SM-G960F 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.IllegalStateException: Fragment BluetoothSettingsFragment{c144f43} (54d161b9-694c-4332-af4c-454dd0040b92)} not attached to a context.
	at androidx.fragment.app.Fragment.requireContext(Fragment.java:805)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onDeviceFound(BluetoothSettingsFragment.java:205)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.access$000(BluetoothSettingsFragment.java:69)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1$1.run(BluetoothSettingsFragment.java:132)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:237)
	at android.app.ActivityThread.main(ActivityThread.java:8167)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)
```