oliexdev/openScale #762: Crash on launch after enabling metrics

tetsuo55, 2021-07-28:
**Describe the bug**
The app crashes on launch when i enable fat free mass , bone mass and visceral fat after measuring without them enabled on my Exingtech Y1

**To Reproduce**
Steps to reproduce the behavior:
1. Launch the app
3. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 2.3.2-dev_5d697fe9_2020-11-05

**Expected behavior**
App launches

**Additional context**
AEG branded Exingtech Y1

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

Build version: 2.3.2-dev_5d697fe9_2020-11-05 
Build date: 1981-01-01 01:01:02 
Current date: 2021-07-28 19:39:40 
Device: Xiaomi Mi 9T Pro 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.ArrayIndexOutOfBoundsException: length=4; index=4
 at com.health.openscale.core.utils.PolynomialFitter.gj_swap(PolynomialFitter.java:186)
 at com.health.openscale.core.utils.PolynomialFitter.gj_echelonize(PolynomialFitter.java:156)
 at com.health.openscale.core.utils.PolynomialFitter.getBestFit(PolynomialFitter.java:118)
 at com.health.openscale.gui.measurement.ChartMeasurementView.addPredictionLine(ChartMeasurementView.java:535)
 at com.health.openscale.gui.measurement.ChartMeasurementView.addTrendLine(ChartMeasurementView.java:512)
 at com.health.openscale.gui.measurement.ChartMeasurementView.refreshMeasurementList(ChartMeasurementView.java:392)
 at com.health.openscale.gui.measurement.ChartMeasurementView.updateMeasurementList(ChartMeasurementView.java:347)
 at com.health.openscale.gui.overview.OverviewFragment.updateOnView(OverviewFragment.java:264)
 at com.health.openscale.gui.overview.OverviewFragment$8.onChanged(OverviewFragment.java:239)
 at com.health.openscale.gui.overview.OverviewFragment$8.onChanged(OverviewFragment.java:236)
 at androidx.lifecycle.LiveData.considerNotify(LiveData.java:131)
 at androidx.lifecycle.LiveData.dispatchingValue(LiveData.java:149)
 at androidx.lifecycle.LiveData.setValue(LiveData.java:307)
 at androidx.lifecycle.LiveData$1.run(LiveData.java:91)
 at android.os.Handler.handleCallback(Handler.java:914)
 at android.os.Handler.dispatchMessage(Handler.java:100)
 at android.os.Looper.loop(Looper.java:224)
 at android.app.ActivityThread.main(ActivityThread.java:7560)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)

oliexdev, 2021-08-22:
please try out the latest dev version, this error should be fixed.

closed 2021-08-22 by oliexdev

tetsuo55, 2021-08-22:
I installed the new dev build over the previous one and the error is the same, i looked at the commit history and can't find any commits that could fix this, is the dev build, built from a different source?

tetsuo55, 2021-08-22:
Build version: 2.3.2-dev_5d697fe9_2020-11-05 
Build date: 1981-01-01 01:01:02 
Current date: 2021-08-22 16:26:54 
Device: Xiaomi Mi 9T Pro 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.ArrayIndexOutOfBoundsException: length=4; index=4
	at com.health.openscale.core.utils.PolynomialFitter.gj_swap(PolynomialFitter.java:186)
	at com.health.openscale.core.utils.PolynomialFitter.gj_echelonize(PolynomialFitter.java:156)
	at com.health.openscale.core.utils.PolynomialFitter.getBestFit(PolynomialFitter.java:118)
	at com.health.openscale.gui.measurement.ChartMeasurementView.addPredictionLine(ChartMeasurementView.java:535)
	at com.health.openscale.gui.measurement.ChartMeasurementView.addTrendLine(ChartMeasurementView.java:512)
	at com.health.openscale.gui.measurement.ChartMeasurementView.refreshMeasurementList(ChartMeasurementView.java:392)
	at com.health.openscale.gui.measurement.ChartMeasurementView.updateMeasurementList(ChartMeasurementView.java:347)
	at com.health.openscale.gui.overview.OverviewFragment.updateOnView(OverviewFragment.java:264)
	at com.health.openscale.gui.overview.OverviewFragment$8.onChanged(OverviewFragment.java:239)
	at com.health.openscale.gui.overview.OverviewFragment$8.onChanged(OverviewFragment.java:236)
	at androidx.lifecycle.LiveData.considerNotify(LiveData.java:131)
	at androidx.lifecycle.LiveData.dispatchingValue(LiveData.java:149)
	at androidx.lifecycle.LiveData.setValue(LiveData.java:307)
	at androidx.lifecycle.LiveData$1.run(LiveData.java:91)
	at android.os.Handler.handleCallback(Handler.java:914)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:224)
	at android.app.ActivityThread.main(ActivityThread.java:7560)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)

tetsuo55, 2021-08-22:
Uninstalled and reinstalled, redownloaded the new APK 3 times, i notice the build version and date hasn't changed compared to the previous build