oliexdev/openScale #763: Huawei Scale 3

mihailstoynov, 2021-08-06:
**Huawei scale 3**
_Write down the scale name and add a link to the vendors app and the product page_

https://consumer.huawei.com/en/accessories/scale-3/
https://play.google.com/store/apps/details?id=com.huawei.health&hl=en_US&gl=US

_To support a new scale it is necessary to gather some information._

**Step 1: Read the general reverse engineer process**

- [X] I have read the [how to reverse engineer a Bluetooth 4.x wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)  entry

**Step 2: Acquiring some Bluetooth traffic**
_Attach 3 log files with the corresponding true values, read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic) for further information._

1. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex male
body height 5ft 11in
DOB 30.07.1981
activity level
weight 209.3lb
```
measured true values in the vendors app for the 1. HCI Snoop log file:
```
date: Tue August 3, 2021 20:09
weight: 212.52 lb
bmi: 29.7
body fat rate: 28.2%
skeletal muscle mass: 82.4 lb
fisceral fat:16.0 level
basal metabolic rate: 1,951 kcal/d
body water: 48.6%
bone mineral content: 7.86 lb
protein: 19.5%
fat-free mass: 152.6 lb
heart rate: 79 bpm
body composition:
- water: 103.33
- protein: 41.40
- fat: 59.92
- bone mineral content: 7.87
```
[2021.08.03.btsnoop_hci.log](https://github.com/oliexdev/openScale/files/6945860/2021.08.03.btsnoop_hci.log)

[Screenshot_20210803-200949738](https://user-images.githubusercontent.com/780207/128523287-349deba7-311a-4ac7-be00-9857baa88eba.jpg)

2. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex male
body height 5ft 11in
DOB 30.07.1981
activity level
weight 209.3lb
```
measured true values in the vendors app for the 2. HCI Snoop log file:
```
date: Tue August 6, 2021 15:15
weight: 214.95 lb
bmi: 30.0
body fat rate: 27.9%
skeletal muscle mass: 83.7 lb
fisceral fat:16.0 level
basal metabolic rate: 1,986 kcal/d
body water: 49.7%
bone mineral content: 7.96 lb
protein: 18.7%
fat-free mass: 155.0 lb
heart rate: 83 bpm
body composition:
- water: 106.90
- protein: 40.12
- fat: 59.97
- bone mineral content: 7.96
```

[2021.08.06.p1.btsnoop_hci.log](https://github.com/oliexdev/openScale/files/6946067/2021.08.06.p1.btsnoop_hci.log)

[Screenshot_20210806-151551362](https://user-images.githubusercontent.com/780207/128523849-c3da0cbd-319f-4bed-b5d4-8b0b40a1b3e7.jpg)

[Screenshot_20210806-151626126](https://user-images.githubusercontent.com/780207/128523856-24dcad43-9100-4dfd-a1c5-b3d9146b82ef.jpg)

3. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex male
body height 5ft 11in
DOB 30.07.1981
activity level
weight 209.3lb
```
measured true values in the vendors app for the 3. HCI Snoop log file:
```
date: Tue August 6, 2021 15:28
weight: 221.89 lb
bmi: 31.0
body fat rate: 29.5%
skeletal muscle mass: 85.7 lb
fisceral fat:17.0 level
basal metabolic rate: 2,039 kcal/d
body water: 48.8%
bone mineral content: 8.11 lb
protein: 18.1%
fat-free mass: 156.5 lb
heart rate: 79 bpm
body composition:
- water: 108.22
- protein: 40.11
- fat: 65.45
- bone mineral content: 8.11
```
[2021.08.06.p2.btsnoop_hci.log](https://github.com/oliexdev/openScale/files/6946095/2021.08.06.p2.btsnoop_hci.log)

[Screenshot_20210806-152839321](https://user-images.githubusercontent.com/780207/128524411-31cab112-b52e-4cf3-a575-84183fc9d3c9.jpg)

[Screenshot_20210806-152903900](https://user-images.githubusercontent.com/780207/128524435-814532ee-c743-46b0-81c2-85b2d6739015.jpg)

4. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex male
body height 5ft 11in
DOB 30.07.1981
activity level
weight 209.3lb
```
measured true values in the vendors app for the 4. HCI Snoop log file:
```
date: Tue August 6, 2021 15:33
weight: 244.71 lb
bmi: 34.2
body fat rate: 33.3%
skeletal muscle mass: 89.6 lb
fisceral fat:21.0 level
basal metabolic rate: 2,143 kcal/d
body water: 45.3%
bone mineral content: 8.45 lb
protein: 17.9%
fat-free mass: 163.1 lb
heart rate: 93 bpm
body composition:
- water: 110.87
- protein: 43.92
- fat: 81.48
- bone mineral content: 8.44
```
[2021.08.06.p3.btsnoop_hci.log](https://github.com/oliexdev/openScale/files/6946098/2021.08.06.p3.btsnoop_hci.log)

[Screenshot_20210806-153400839](https://user-images.githubusercontent.com/780207/128524925-8b9a8a58-5475-4413-9b1a-d4539eeab8ab.jpg)

[Screenshot_20210806-153334131](https://user-images.githubusercontent.com/780207/128524957-96e302a5-17a7-4f07-8524-258ff4d68a8b.jpg)

45 Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex male
body height 5ft 11in
DOB 30.07.1981
activity level
weight 209.3lb
```
measured true values in the vendors app for the 5. HCI Snoop log file:
```
date: Tue August 6, 2021 15:37
weight: 209.33 lb
bmi: 29.3
body fat rate: 27.0%
skeletal muscle mass: 83.6 lb
fisceral fat:15.0 level
basal metabolic rate: 1,983 kcal/d
body water: 51.1%
bone mineral content: 7.92 lb
protein: 18.1%
fat-free mass: 152.8 lb
heart rate: 80 bpm
body composition:
- water: 106.97
- protein: 37.92
- fat: 56.53
- bone mineral content: 7.91
```
[2021.08.06.p4.btsnoop_hci.log](https://github.com/oliexdev/openScale/files/6946125/2021.08.06.p4.btsnoop_hci.log)

[Screenshot_20210806-153753996](https://user-images.githubusercontent.com/780207/128525568-7e53e8f8-da47-42b2-8e65-46d603c85230.jpg)

[Screenshot_20210806-153821429](https://user-images.githubusercontent.com/780207/128525578-3c50091d-5057-4e97-9997-95ef85ceb7fa.jpg)

**Step 3: Discover Bluetooth services and characteristic**
_Read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#2-find-out-the-bluetooth-services-and-characteristic) how to create the openScale debug file_

[openScale_2021-08-06_15-44.txt](https://github.com/oliexdev/openScale/files/6946129/openScale_2021-08-06_15-44.txt)

aubergene, 2022-04-01:
Thanks for capturing this. Any ideas what the next steps would be to add support for this scale?

yabdali, 2022-06-25:
Hi, I am intending to buy this scale.  Is their a plan to support it in the near future?  Appreciate your feedback, thanks

patoroco, 2023-02-07:
Same question here. I’m interested on the support for this model