oliexdev/openScale #766: find ble page list can't scroll

mbsg, 2021-08-17:
my phone find more than one page ble device, but my scale device at more bottom postion, I can't choose my scale device
[image]

oliexdev, 2021-08-21:
try out the latest dev version at https://github.com/oliexdev/openScale/releases/download/travis-dev-build/openScale-dev-build.apk

mbsg, 2021-09-02:
[image]
my device still can't scroll list. my phone is oneplus 7 pro using hydrogen 11.0.2.1 gm21 os

oliexdev, 2021-10-10:
could you post the openScale log file please and please use the latest openScale dev version.