oliexdev/openScale #769: Can't find RENPHO ES-CS20M

mikewilzn, 2021-08-21:
Has anyone had success pairing this scale? I've read the FAQ and nothing on there is the issue. It has never been connected to my phone at all. I don't really want to have to make an account with Renpho to use their app.

oliexdev, 2021-08-22:
what is shown if you search for the scale in openScale? Could you attach a screenshot?

guix77, 2021-09-02:
From what I understand, this scale works differently than other bluetooth scales. You do not pair it with your phone, and it's not even visible. In fact, it only activates bluetooth when someone is on the scale. That's why you have to be on the scale to add it to the Renpho app. Notice that it won't pair it with your phone. Instead of pairing, this scale sends the data in the bluetooth discovery protocol, if I understood well.

It's a bad design decision from a standards point of view, however it makes sense from an energy point of view: bluetooth should not be used except when someone is on the scale, which would lead to longer battery life.

That's why you can't see it in OpenScale. I guess that adding this scale in OpenScale would require a lot of work since OpenScales seems to rely on a classic bluetooth pairing + services.

Some other scales are clones (or this one is a clone) and there are some efforts to reverse-engineer the protocol, but I'm not sure we'll see something usable in a near future.

lc-guy, 2021-11-24:
Please reopen this, as the issue isn't related to whether or not it's paired with other apps -- the scale itself isn't discoverable, and most likely gets paired with based on a hardcoded MAC address in the official vendor apps.
One way to fix this would be to allow users to manually enter the MAC address for the scale.

GarnetSunset, 2022-09-04:
Total noob here, so please bare with me. I think I have one of those "it broadcasts but doesn't pair" and I can't find it in the debug log, or in my bluetooth menu, especially so because I don't see it in the bluetooth menu (yes I'm standing on it and it's in the same setting as what the real app supports, so it should be broadcasting)

```
2022-09-04 11:44:47.491 Debug [2] AboutPreferences: Debug log enabled, openScale v2.4.6 (61), SDK 31, OnePlus NE2213
2022-09-04 11:44:47.493 Debug [2] AboutPreferences: Selected user id(0) name() birthday(Sun Sep 04 11:44:47 EDT 2022) age(0) body height(-1.00) scale unit(kg) gender(male) initial weight(-1.00) goal weight(-1.00) goal date(Sun Sep 04 11:44:47 EDT 2022) measure unt(cm) activity level(0) assisted weighing(false)
2022-09-04 11:46:24.031 Debug [2] PermissionHelper: SDK >= 31 request for Bluetooth Scan and Bluetooth connect permissions
2022-09-04 11:46:25.025 Debug [2] BluetoothSettingsFragment: All Bluetooth permissions granted
2022-09-04 11:46:25.052 Info [2] BluetoothCentralManager: scan started
2022-09-04 11:46:25.056 Debug [2] PermissionHelper: SDK >= 31 request for Bluetooth Scan and Bluetooth connect permissions
2022-09-04 11:46:25.076 Debug [2] BluetoothSettingsFragment: All Bluetooth permissions granted
2022-09-04 11:46:25.089 Info [2] BluetoothCentralManager: scan started
2022-09-04 11:46:25.117 Debug [2] BluetoothSettingsFragment: Found unsupported device LHB-31FAA80C [F9:B2:58:50:93:93]
2022-09-04 11:46:25.134 Debug [2] BluetoothSettingsFragment: Found unsupported device E2B-US-NKA0396A [C8:FF:77:20:E5:40]
2022-09-04 11:46:25.149 Debug [2] BluetoothSettingsFragment: Found unsupported device LHB-550428BC [E9:5A:98:4D:EA:91]
2022-09-04 11:46:25.161 Debug [2] BluetoothSettingsFragment: Found unsupported device [TV] Absolem [8C:79:F5:EC:43:B3]
2022-09-04 11:46:25.275 Debug [2] BluetoothSettingsFragment: Found unsupported device [TV] Samsung 6 Series (65) [D8:E0:E1:3F:04:67]
2022-09-04 11:46:25.528 Debug [2] BluetoothSettingsFragment: Found unsupported device [LG] webOS TV OLED55C1AUB [75:E3:CC:94:E7:23]
2022-09-04 11:46:25.544 Debug [2] BluetoothSettingsFragment: Found unsupported device [TV] Lily Cocapina [64:E7:D8:68:ED:E5]
2022-09-04 11:46:25.799 Debug [2] BluetoothSettingsFragment: Found unsupported device Living Room [44:C6:5D:7B:ED:6D]
2022-09-04 11:46:26.214 Debug [2] BluetoothSettingsFragment: Found unsupported device Irie Bella  [CD:7A:77:9C:02:75]
2022-09-04 11:46:26.450 Debug [2] BluetoothSettingsFragment: Found unsupported device SHIELD [58:7C:78:17:63:C0]
2022-09-04 11:46:27.588 Debug [2] BluetoothSettingsFragment: Found unsupported device 1F02C427AB6B [78:6D:EB:29:87:F9]
2022-09-04 11:46:32.993 Debug [2] BluetoothSettingsFragment: Found unsupported device TG1-CFC638162B19 [CF:C6:38:16:2B:19]
2022-09-04 11:46:33.063 Debug [2] BluetoothSettingsFragment: Found unsupported device Bedroom TV [53:1E:98:43:A3:B3]
2022-09-04 11:46:34.355 Debug [2] BluetoothSettingsFragment: Found unsupported device Venus_2C4CC603CA8F [2C:4C:C6:03:CA:8F]
2022-09-04 11:46:38.372 Debug [2] BluetoothSettingsFragment: Found unsupported device TG1-EF28C58D5CDD [EF:28:C5:8D:5C:DD]
2022-09-04 11:46:39.441 Debug [2] BluetoothSettingsFragment: Found unsupported device Nanoleaf A19 2NNA [CE:3F:B6:D8:E1:C6]
2022-09-04 11:46:39.959 Debug [2] BluetoothSettingsFragment: Found unsupported device Nanoleaf A19 7WFW [CC:AB:13:C1:95:8A]
2022-09-04 11:46:40.005 Debug [2] BluetoothSettingsFragment: Found unsupported device [ AV ] Samsung Soundbar Q70R [24:FC:E5:AE:4D:47]
2022-09-04 11:46:40.452 Debug [2] BluetoothSettingsFragment: Found unsupported device LE-Bose Free SoundSport [2C:41:A1:D8:72:5A]
2022-09-04 11:46:40.919 Debug [2] BluetoothSettingsFragment: Found unsupported device [TV] Samsung 7 Series (55) [BC:7E:8B:24:61:0D]
2022-09-04 11:46:45.056 Info [2] BluetoothCentralManager: scan stopped
2022-09-04 11:46:50.410 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2022-09-04 11:46:50.700 Debug [2] BluetoothSettingsFragment: Found unsupported device GoPlug [11:22:30:00:00:51]
2022-09-04 11:46:51.303 Debug [2] PermissionHelper: SDK >= 31 request for Bluetooth Scan and Bluetooth connect permissions
2022-09-04 11:46:51.329 Debug [2] BluetoothSettingsFragment: All Bluetooth permissions granted
2022-09-04 11:46:51.336 Info [2] BluetoothCentralManager: scan started
2022-09-04 11:46:51.436 Debug [2] BluetoothSettingsFragment: Found unsupported device Living Room [44:C6:5D:7B:ED:6D]
2022-09-04 11:46:51.453 Debug [2] BluetoothSettingsFragment: Found unsupported device LHB-31FAA80C [F9:B2:58:50:93:93]
2022-09-04 11:46:51.466 Debug [2] BluetoothSettingsFragment: Found unsupported device LHB-550428BC [E9:5A:98:4D:EA:91]
2022-09-04 11:46:51.484 Debug [2] BluetoothSettingsFragment: Found unsupported device [TV] Absolem [8C:79:F5:EC:43:B3]
2022-09-04 11:46:51.500 Debug [2] BluetoothSettingsFragment: Found unsupported device TG1-CFC638162B19 [CF:C6:38:16:2B:19]
2022-09-04 11:46:51.519 Debug [2] BluetoothSettingsFragment: Found unsupported device E2B-US-NKA0396A [C8:FF:77:20:E5:40]
2022-09-04 11:46:51.621 Debug [2] BluetoothSettingsFragment: Found unsupported device [LG] webOS TV OLED55C1AUB [75:E3:CC:94:E7:23]
2022-09-04 11:46:51.670 Debug [2] BluetoothSettingsFragment: Found unsupported device Irie Bella  [CD:7A:77:9C:02:75]
2022-09-04 11:46:52.759 Debug [2] BluetoothSettingsFragment: Found unsupported device SHIELD [58:7C:78:17:63:C0]
[... cut ...]
2022-09-04 11:49:07.198 Debug [2] BluetoothSettingsFragment: Found unsupported device GoPlug [11:22:30:00:00:51]
2022-09-04 11:49:25.044 Debug [2] BluetoothCentralManager: scanning timeout, restarting scan
2022-09-04 11:49:25.050 Info [2] BluetoothCentralManager: scan stopped
2022-09-04 11:49:26.075 Info [2] BluetoothCentralManager: scan started
2022-09-04 11:50:10.146 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2022-09-04 11:50:13.015 Debug [2] AboutPreferences: Debug log disabled
```

GraciousGpal, 2022-09-17:
I recently bought one of these devices and I figured out how to extract weight from them.
I couldn't really confirm other metrics but here is the implementation. Note: I tested this on a PC that had a Bluetooth adapter. I did not have more time to investigative other metrics.
```py
import asyncio
from bleak import BleakScanner

class EsCS20M:
    """
    This class gets the weight value from Renpho ES-CS20M
    I got other properties but did not have time to confirm so they are not included.
    """

    def __init__(self, mac: str, array: bytearray) -> None:
        self.mac = mac
        self.array = array

    def __repr__(self):
        return f"{self.__class__.__name__}(mac={self.mac}, array={self.array})"

    @property
    def weight(self) -> float:
        """
        Returns 16 bit register value and converts it into weight.
        """
        return ((self.array[18] << 8) | self.array[17])/100

class MyScanner:
    """
    Basic class implementation to demonstrate how to extract data from broadcast type scales.
    """
    def __init__(self, device_mac) -> None:
        self.scanning = asyncio.Event()
        self._scanner = BleakScanner()
        self._scanner.register_detection_callback(self.detection_callback)
        self.device_mac = device_mac

    def detection_callback(self, device, advertisement_data) -> None:
        """
        Called when bleak scanner get advertisment_data 
        """
        if device.address == self.device_mac:
            # 65535 just came from my device could be different for you.
            _array = bytearray(advertisement_data.manufacturer_data[65535])
            c60_data = EsCS20M(mac=self.device_mac, array=_array)
            print(f"data: {c60_data} weight: {c60_data.weight}")

    async def run(self) -> None:
        await self._scanner.start()
        self.scanning.set()
        while self.scanning.is_set():
            await asyncio.sleep(0.1)
        await self._scanner.stop()

if __name__ == '__main__':
    my_device_mac_address = "ED:67:3B:17:E4:5F"
    my_scanner = MyScanner(my_device_mac_address)
    loop = asyncio.get_event_loop()
    loop.run_until_complete(my_scanner.run())
```

lc-guy, 2022-09-22:
@oliexdev reopen please?

ShockStricken, 2024-12-05:
Any updates on this? I'm experiencing the same, scale works fine in the Renpho Health app but is not discoverable at all in OpenScale.