oliexdev/openScale #771: crash when connecting to QN-Scale

Hoernchen, 2021-08-30:
Happens with the release version as well as with the current master immediately upon connecting, app is therefore unusable:

    D/BluetoothCommunication: Found peripheral 'QN-Scale'
    D/BluetoothAdapter: isLeEnabled(): ON
    I/BluetoothCentral: scan stopped
    D/BluetoothCommunication: Try to connect to BLE device D8:0B:CB:14:6C:E7
    I/BluetoothPeripheral: connect to 'QN-Scale' (D8:0B:CB:14:6C:E7) using TRANSPORT_LE
    D/BluetoothGatt: connect() - device: D8:0B:CB:14:6C:E7, auto: false
    D/BluetoothGatt: registerApp()
    D/BluetoothGatt: registerApp() - UUID=b0174713-c339-4b9b-b0c4-1988b0dfd814
    D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
    D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=D8:0B:CB:14:6C:E7
    I/BluetoothPeripheral: connected to 'QN-Scale' (BOND_NONE) in 0,2s
    D/BluetoothPeripheral: discovering services of 'QN-Scale' with delay of 0 ms
    D/BluetoothGatt: discoverServices() - device: D8:0B:CB:14:6C:E7
    D/BluetoothGatt: onConnectionUpdated() - Device=D8:0B:CB:14:6C:E7 interval=6 latency=0 timeout=500 status=0
    D/BluetoothGatt: onSearchComplete() = Device=D8:0B:CB:14:6C:E7 Status=0
    I/BluetoothPeripheral: discovered 5 services for 'QN-Scale'
    D/BluetoothCommunication: connected to 'QN-Scale'
    D/BluetoothCommunication: Successful Bluetooth services discovered
    D/BluetoothCommunication: Resume machine state
    D/BluetoothCommunication: Step Nr 0
    D/BluetoothCommunication: Invoke set notification on 0xffe1
    D/BluetoothCommunication: Step Nr 1
    D/BluetoothCommunication: Invoke set indication on 0xffe2
    D/BluetoothCommunication: Step Nr 2
    D/BluetoothCommunication: Invoke write bytes [13 09 15 01 10 00 00 00 42] on 0xffe3
    D/AndroidRuntime: Shutting down VM
    E/AndroidRuntime: FATAL EXCEPTION: main
        Process: com.health.openscale, PID: 26816
        java.lang.NullPointerException: no valid characteristic provided
            at java.util.Objects.requireNonNull(Objects.java:245)
            at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1182)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:243)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:231)
            at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:503)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:505)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:505)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:173)
            at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:372)
            at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:323)
            at android.os.Handler.handleCallback(Handler.java:938)
            at android.os.Handler.dispatchMessage(Handler.java:99)
            at android.os.Looper.loop(Looper.java:223)
            at android.app.ActivityThread.main(ActivityThread.java:7656)
            at java.lang.reflect.Method.invoke(Native Method)
            at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
    E/CustomActivityOnCrash: App has crashed, executing CustomActivityOnCrash's UncaughtExceptionHandler
        java.lang.NullPointerException: no valid characteristic provided
            at java.util.Objects.requireNonNull(Objects.java:245)
            at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1182)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:243)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:231)
            at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:503)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:505)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:505)
            at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:173)
            at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:372)
            at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:323)
            at android.os.Handler.handleCallback(Handler.java:938)
            at android.os.Handler.dispatchMessage(Handler.java:99)
            at android.os.Looper.loop(Looper.java:223)
            at android.app.ActivityThread.main(ActivityThread.java:7656)
            at java.lang.reflect.Method.invoke(Native Method)
            at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
    D/BluetoothGatt: onConnectionUpdated() - Device=D8:0B:CB:14:6C:E7 interval=36 latency=0 timeout=500 status=0
    I/Process: Sending signal. PID: 26816 SIG: 9
    Disconnected from the target VM, address: 'localhost:32997', transport: 'socket'

Hoernchen, 2021-08-31:
I'm working on this and I've fixed it for my scale, I just have to add a way to distinguish the scales by looking at the scale config, since there are apparently a dozen different "qn scale" models out there.

gammerboy, 2021-09-28:
Hi. Seems I'm having the same issue.
Here are the logs I grabbed from the app in case it helps but not sure if this is helpful.

```
Build version: 2.3.2-dev_5d697fe9_2020-11-05 
Build date: 1981-01-01 01:01:02 
Current date: 2021-09-28 19:00:38 
Device: OnePlus ONEPLUS A3003 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:245)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1192)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:200)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:188)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:460)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:462)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:462)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:173)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:329)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:381)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:223)
	at android.app.ActivityThread.main(ActivityThread.java:7664)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
```

gammerboy, 2021-09-28:
The scale I'm trying to connect to is (https://github.com/oliexdev/openScale/issues/373) exactly like the one shown in that issue. But it's called Actofit smartscale Lite.

oliexdev, 2021-10-11:
@gammerboy try out the openScale version 2.4.1

gammerboy, 2021-10-12:
Hi,
Here is the log from 2.4.1. It still crashes while connecting to the scale.

```
Build version: 2.4.1 
Build date: 1981-01-01 01:01:02 
Current date: 2021-10-12 21:15:56 
Device: Samsung SM-G780F 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:245)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1182)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:598)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:600)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:600)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:209)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:246)
	at android.app.ActivityThread.main(ActivityThread.java:8595)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
```

imcan8dn, 2021-10-17:
Crashes when connecting to QN-Scale1.

Log:

Build version: 2.4.1 
Build date: 1981-01-01 01:01:02 
Current date: 2021-10-16 21:22:17 
Device: Google Pixel 3a XL 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:245)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1182)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:598)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:600)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:600)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:209)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:223)
	at android.app.ActivityThread.main(ActivityThread.java:7664)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)

http403, 2022-02-14:
I got similar error when reading data from a Yolanda CS20E.
```
Build version: 2.4.5 
Build date: 1981-01-01 01:01:02 
Current date: 2022-02-14 16:52:53 
Device: Sony H8324 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:228)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1206)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:209)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:359)
	at android.app.ActivityThread.main(ActivityThread.java:7418)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
```

sethbaird01, 2022-08-24:
Same error here, QN-Scale (Vont SBS01)

```Build version: 2.4.6 Build date: 1981-01-01 01:01:02 Current date: 2022-08-24 18:48:30 Device: Google Pixel 6 OS version: Android 12 (SDK 32)  Stack trace:  java.lang.NullPointerException: no valid characteristic provided	at java.util.Objects.requireNonNull(Objects.java:245)	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1202)	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)	at android.os.Handler.handleCallback(Handler.java:938)	at android.os.Handler.dispatchMessage(Handler.java:99)	at android.os.Looper.loopOnce(Looper.java:201)	at android.os.Looper.loop(Looper.java:288)	at android.app.ActivityThread.main(ActivityThread.java:7870)	at java.lang.reflect.Method.invoke(Native Method)	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)```

bjbdev, 2022-10-28:
Same issue here with scales branded Etekcity ESF24.

```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2022-10-28 10:50:41 
Device: Google Pixel 5 
OS version: Android 13 (SDK 33) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
 at java.util.Objects.requireNonNull(Objects.java:245)
 at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1202)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
 at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
 at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
 at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
 at android.os.Handler.handleCallback(Handler.java:942)
 at android.os.Handler.dispatchMessage(Handler.java:99)
 at android.os.Looper.loopOnce(Looper.java:201)
 at android.os.Looper.loop(Looper.java:288)
 at android.app.ActivityThread.main(ActivityThread.java:7898)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
```

Some BLE attribute inspection in case that helps give a way to differentiate the scales:
```
handle: 0x0002, char properties: 0x02, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0006, char properties: 0x02, char value handle: 0x0007, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x0009, char properties: 0x20, char value handle: 0x000a, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x000d, char properties: 0x02, char value handle: 0x000e, uuid: 00002a19-0000-1000-8000-00805f9b34fb
handle: 0x0010, char properties: 0x10, char value handle: 0x0011, uuid: 0000fff1-0000-1000-8000-00805f9b34fb
handle: 0x0013, char properties: 0x08, char value handle: 0x0014, uuid: 0000fff2-0000-1000-8000-00805f9b34fb
handle: 0x0016, char properties: 0x06, char value handle: 0x0017, uuid: 00010203-0405-0607-0809-0a0b0c0d2b12

handle: 0x0002, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0006, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0007, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x0008, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x0009, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x000a, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x000b, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x000c, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x000d, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x000e, uuid: 00002a19-0000-1000-8000-00805f9b34fb
handle: 0x000f, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x0010, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0011, uuid: 0000fff1-0000-1000-8000-00805f9b34fb
handle: 0x0012, uuid: 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0013, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0014, uuid: 0000fff2-0000-1000-8000-00805f9b34fb
handle: 0x0015, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x0016, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0017, uuid: 00010203-0405-0607-0809-0a0b0c0d2b12
handle: 0x0018, uuid: 00002901-0000-1000-8000-00805f9b34fb
```

```
> 0x0001 Characteristic value/descriptor: 00 18
> 0x0002 Characteristic value/descriptor: 02
> 0x0003 Characteristic value/descriptor: 51 4e 2d 53 63 61 6c 65 00 00 00 00 00 00 00 00 00 00
> 0x0004 Characteristic value/descriptor: 02
> 0x0005 Characteristic value/descriptor: 00 00
> 0x0006 Characteristic value/descriptor: 02
> 0x0007 Characteristic value/descriptor: 14 00 50 00 00 00 90 01
> 0x0008 Characteristic value/descriptor: 01 18
> 0x0009 Characteristic value/descriptor: 20
> 0x000a Characteristic value/descriptor: 00 00 00 00
> 0x000b Characteristic value/descriptor: 00 00
> 0x000c Characteristic value/descriptor: 0f 18
> 0x000d Characteristic value/descriptor: 02
> 0x000e Characteristic value/descriptor: 53
> 0x000f Characteristic value/descriptor: f0 ff
> 0x0010 Characteristic value/descriptor: 10
> 0x0011 Characteristic value/descriptor: 12 0f 15 fd 4f 26 cb 0b d8 30 05 30 00 01 bc 00 00 00 00 00
> 0x0012 Characteristic value/descriptor: 00 00
> 0x0013 Characteristic value/descriptor: 08
> 0x0015 Characteristic value/descriptor: 12 19 0d 0c 0b 0a 09 08 07 06 05 04 03 02 01 00
> 0x0016 Characteristic value/descriptor: 06
> 0x0017 Characteristic value/descriptor: 00
> 0x0018 Characteristic value/descriptor: 4f 54 41
```

SuperPrower, 2022-12-23:
```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2022-12-23 07:38:40 
Device: Google Pixel 5 
OS version: Android 13 (SDK 33) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
 at java.util.Objects.requireNonNull(Objects.java:245)
 at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1202)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
 at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
 at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)
 at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
 at android.os.Handler.handleCallback(Handler.java:942)
 at android.os.Handler.dispatchMessage(Handler.java:99)
 at android.os.Looper.loopOnce(Looper.java:201)
 at android.os.Looper.loop(Looper.java:288)
 at android.app.ActivityThread.main(ActivityThread.java:7898)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
 ```
 
Renpho ES-26M-W/B

DuckDuckWhale, 2023-01-15:
Renpho ES-CS20M is listed as a supported device yet crashes in the same way:
```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2023-01-15 13:41:33 
Device: Google Pixel 7 
OS version: Android 13 (SDK 33) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:245)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1202)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7872)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
	
```

oliexdev, 2023-01-22:
> I'm working on this and I've fixed it for my scale, I just have to add a way to distinguish the scales by looking at the scale config, since there are apparently a dozen different "qn scale" models out there.

@Hoernchen how did you fixed it?

andreaspeters, 2023-01-30:
> Renpho ES-CS20M is listed as a supported device yet crashes in the same way:
>
> ```
> Build version: 2.4.6
> Build date: 1981-01-01 01:01:02
> Current date: 2023-01-15 13:41:33
> Device: Google Pixel 7
> OS [...]

Same here

anshul-g-AnAr, 2023-07-17:
has anyone found the way? I'm  also having QN-Scale but it's payload format or byte structure is unknown to me and perhaps may be different from this repo, that might be the issue. if any one know how to know the byte format please help.

g4rb4g3, 2023-08-14:
I ran into the same issue on a Renpho ES-CS20M

```
Build version: 2.5.2 
Build date: 1981-01-01 01:01:02 
Current date: 2023-08-14 08:01:06 
Device: Samsung SM-S911B 
OS version: Android 13 (SDK 33) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1229)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8757)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)

```

eliasp, 2024-06-24:
Another `Renpho ES-CS20M` crash:

```
Build version: 2.5.2 
Build date: 1981-01-01 01:01:02 
Current date: 2024-06-24 18:05:19 
Device: OnePlus AC2003 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1229)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:245)
	at android.app.ActivityThread.main(ActivityThread.java:8004)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:631)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:978)
```

weliem, 2024-06-24:
the error is very clear: the app is trying to write to a characteristic that the scale doesn't have. Apparently, not all QN scales have the same characteristics....

behrisch, 2025-02-02:
I have the same error here with an Arboleaf CS20G which identifies itself as QN-Scale in the bluetooth logs. Anything I can do to help?

eliasp, 2025-09-09:
I can report, that the `Renpho ES-CS20M` doesn't crash the app anymore!
The only downside: some values are off - weight and BMI are divided by 10 (e.g. 7.8kg instead of 78kg)

RobertKillick, 2025-09-10:
The scale is branded `Etekcity` model `ESF24`

```
Build version: 2.5.4
Build date: 1981-01-01 01:01:02
Current date: 2025-09-10 12:29:03
Device: Samsung SM-S908B
OS version: Android 15 (SDK 35)

Stack trace:
  java.lang.IllegalArgumentException: characteristic <0000fff2-0000-1000-8000-00805f9b34fb> does not support writeType 'WITH_RESPONSE'
  at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1238)
  at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
  at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
  at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:136)
  at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
  at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
  at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onNotificationStateUpdate(BluetoothCommunication.java:472)
  at com.welie.blessed.BluetoothPeripheral$1$2.run(BluetoothPeripheral.java:244)
  at android.os.Handler.handleCallback(Handler.java:959)
  at android.os.Handler.dispatchMessage(Handler.java:100)
  at android.os.Looper.loopOnce(Looper.java:257)
  at android.os.Looper.loop(Looper.java:342)
  at android.app.ActivityThread.main(ActivityThread.java:9634)
  at java.lang.reflect.Method.invoke(Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:619)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:929)
`````