oliexdev/openScale #772: Eufy Smart Scale T9140

unlimitedbacon, 2021-09-11:
**Eufy Smart Scale T9140**
[Product Page](https://us.eufylife.com/products/t9140)
[EufyLife (Vendor App)](https://play.google.com/store/apps/details?id=com.oceanwing.smarthome&hl=en_US&gl=US)

**Step 1: Read the general reverse engineer process**

- [x] I have read the [how to reverse engineer a Bluetooth 4.x wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)  entry

**Step 2: Acquiring some Bluetooth traffic**
_Attach 3 log files with the corresponding true values, read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic) for further information._

1. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
Sex: male
Body Height: 6'3"
Age: 30
Activity Level: N/A
```
measured true values in the vendors app for the 1. HCI Snoop log file:
```
Weight: 76.0 kg
Date: Sep 11 2021, 2:52 PM
BMI: 20.8
Body Fat %: 14.1%
Water: 62.7%
Muscle Mass %: 44.5%
Bome Mass %: 8.4%
BMR: 2857.6 kcal
Visceral Fat: 3.4
Lean Body Mass: 65.3 kg
Body Fat Mass: 10.7 kg
Bone Mass: 6.4 kg
Muscle Mass: 33.8 kg
```
_--> Attach the 1. HCI Snoop log file here <--_
[hci_snoop_2021_09_11_14_52_23.cfa.zip](https://github.com/oliexdev/openScale/files/7148620/hci_snoop_2021_09_11_14_52_23.cfa.zip)

2. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
Sex: male
Body Height: 6'3"
Age: 30
Activity Level: N/A
```
measured true values in the vendors app for the 2. HCI Snoop log file:
```
Weight: 79.9 kg
Date: Sep 11 2021, 3:53 PM
BMI: 21.9
Body Fat %: 16.4%
Water: 61.0%
Muscle Mass %: 43.5%
Bome Mass %: 8.4%
BMR: 2919.6 kcal
Visceral Fat: 4.9
Lean Body Mass: 66.8 kg
Body Fat Mass: 13.1 kg
Bone Mass: 6.7 kg
Muscle Mass: 34.8 kg
```

_--> Attach the 2. HCI Snoop log file here <--_
[hci_snoop_2021_09_11_15_52_14.cfa.zip](https://github.com/oliexdev/openScale/files/7148633/hci_snoop_2021_09_11_15_52_14.cfa.zip)

3. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
Sex: male
Body Height: 6'3"
Age: 30
Activity Level: N/A
```
measured true values in the vendors app for the 3. HCI Snoop log file:
```
Weight: 84.0 kg
Date: Sep 11 2021, 4:23 PM
BMI: 23.0
Body Fat %: 18.7%
Water: 59.4%
Muscle Mass %: 42.6%
Bone Mass %: 8.5%
BMR: 2984.8 kcal
Visceral Fat: 10.2
Lean Body Mass: 68.3 kg
Body Fat Mass: 15.7 kg
Bone Mass: 7.1 kg
Muscle Mass: 35.8 kg
```
_--> Attach the 3. HCI Snoop log file here <--_
[hci_snoop_2021_09_11_16_22_11.cfa.zip](https://github.com/oliexdev/openScale/files/7148654/hci_snoop_2021_09_11_16_22_11.cfa.zip)

**Step 3: Discover Bluetooth services and characteristic**
_Read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#2-find-out-the-bluetooth-services-and-characteristic) how to create the openScale debug file_

_--> Attach the openScale debug log file here <--_
[openScale_2021-09-11_16-27.txt](https://github.com/oliexdev/openScale/files/7148665/openScale_2021-09-11_16-27.txt)

Quintasan, 2021-12-03:
Not sure if it's the same model, but I have T9146

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/7651327/btsnoop_hci.log)

[openScale_2021-12-03_18-04.txt](https://github.com/oliexdev/openScale/files/7651328/openScale_2021-12-03_18-04.txt)

Can't say I could find my weight anywhere. I did look for the following values:

| True value | Hexadecimal |
|------------|-----------------|
| 78.9 | 0315 |
| 40.4 | 0194 |
| 38.6 | 0182 |
| 21.9 | 00DB |

MichelEhmen, 2022-05-26:
Are there any updates on this?

Kreuger, 2022-09-05:
I have the T9147. I havent been able to get the snooping to work. But it shows up in the app saying its not supported. And I am able to see it with some details in my BLE Scan app.

qimmiq, 2022-11-17:
i also tried to look through the log for my T9147 but wasn't able to find the weight value

will add files in a sec

cbondurant, 2023-10-26:
I happened to do some reverse engineering work on this particular model a while back, unfortunately I cannot remember exactly where all my scattered notes are just at this moment.

What I was able to dig up was this https://git.sr.ht/~cbondurant/eufy-protocol-reversal/tree/main/item/src/main.rs attempt at connecting and reading the data off of the device live as it is activated. 

I can do some more digging to try and recover some of the other things I had figured out if there might be someone who is interested in doing the implementation for openScale. I don't think I have the android or java knowledge needed to do so myself.

Main thing I can remember now was that line 35 of the linked code printed out the weight when I was running tests back then.

Kreuger, 2024-12-28:
@cbondurant Hey are you still working on this? The page appears dead.

cbondurant, 2024-12-29:
My interest in reversing the scale ended up waning,  I might get back to it at some point? but I've not got any plans to in the near future.

tomdevelops, 2024-12-29:
Hey @cbondurant, unfortunately the link to the code appears to be broken or inaccessible. Would it be possible for you to share your snippet again? That would be incredibly helpful for anyone looking to pick up this work, including myself. Appreciate your contributions to this 🙏

cbondurant, 2024-12-29:
It seems to load for me even when I am logged out of source hut? I can possibly also mirror the code here on github if that might help with access, but the link isn't broken for me.

Kreuger, 2024-12-30:
I had trouble trying to get around most pages on SH. I dont know if they dont like my VPN or what. Id appreciate a mirror here.

cbondurant, 2024-12-30:
Mirrored it to here - https://github.com/cbondurant/eufy-protocol-reversal hope it helps!

Kreuger, 2024-12-30:
Thanks. I made a fork for backup purposes as I am not a dev.

tomdevelops, 2024-12-30:
@cbondurant thx :pray:

Kreuger, 2025-03-04:
> [@cbondurant](https://github.com/cbondurant) thx 🙏

Hey have you gotten anywhere with this?

KevinSJ, 2026-01-08:
I've created the PR and build an [APK](https://github.com/KevinSJ/openScale/releases/tag/beta-build) with [GitHub action](https://github.com/KevinSJ/openScale/actions/runs/20800566634) using [the branch](https://github.com/KevinSJ/openScale/tree/feat/eufy-c1-p1-support) in my fork, if you own C1 or P1 scale, feel free to try and let me know if it works.

KruyKaze, 2026-01-08:
It works on my C1 🎉