oliexdev/openScale #773: implementation of SBSI

jobs62, 2021-09-15:
**Is your feature request related to a problem? Please describe.**
BMI is not always a good fit for diagnosing obesity, partuculary for tall ans strong people.

**Describe the solution you'd like**
Surface-Based Body Shape Index seam to be a better fit for heart related issue prediction. it would require the app to at least propose to save measurement of vertical trunk circumference.

**Additional context**
https://doi.org/10.1371/journal.pone.0144639
https://doi.org/10.32394/rpzh.2019.0077

oliexdev, 2021-10-10:
You posted two different papers with for me exotic formulas. One for Surface-Based Body Shape Index (SBSI) and body shape index (ABSI).
Could you post the formula which you have in mind with the need variables.

gitmuti, 2022-08-28:
I attached an implementation in MS Excel in order to support you. 
[ABSI_and_SBSI_calculation.xlsx](https://github.com/oliexdev/openScale/files/9439612/ABSI_and_SBSI_calculation.xlsx)
May I asked you to add 
- ABSI z  score   (i.e. the normalized ABSI value),
- ABSI Mortality Risk (both can be calculated without a new input measurement) ,
- Vertical Trunk Circumference VTC (cm)  and
- SBSI
Note: The evaluation of SBSI is against fixed values (male/female) +/- Standard Deviation